Pho3niX90/solis_modbus #150: [Bug]: time.solis_s6_solis_time_charging_charge_start_slot_x unavailable

charliegadget, 2025-02-25:
## 🐛 Bug Description

First "bug" report so bear with me!

after upgrading from 3.0.1 to 3.0.4 the edittable windows where you can edit the charge times show as "unavailable" and there for uneditable.

time.solis_s6_solis_time_charging_charge_start_slot_1
time.solis_s6_solis_time_charging_charge_start_slot_2

etc

Discharge is the same.

In 3.0.1 you are able to edit the time in a time window.

Arnoudd, 2025-02-25:
Same here so i reverted to 3.0.2

Arnoudd, 2025-02-25:
There's a change in time.py in version 3.04. I'm not sure why this is changed?

The difference lies in the logic of the conditions, specifically in how the last two checks are combined.
	1.	First code:

if hour is not None and minute is not None and 0 <= minute <= 59 and 0 <= hour <= 23:

	•	This checks that:
	•	Both hour and minute are not None.
	•	minute is between 0 and 59 (inclusive).
	•	hour is between 0 and 23 (inclusive).
	•	This means the code is executed only when hour and minute represent valid time values.

	2.	Second code:

if hour is not None and minute is not None and not (0 <= minute <= 59) and not (0 <= hour <= 23):

	•	This checks that:
	•	Both hour and minute are not None.
	•	minute is not between 0 and 59.
	•	hour is not between 0 and 23.
	•	This means the code is executed only when both hour and minute fall outside their valid ranges.

Summary:
	•	The first code verifies that hour and minute are within their valid ranges.
	•	The second code verifies that hour and minute are outside their valid ranges.

charliegadget, 2025-02-25:
Yes, I saw that too and wondered if it had something to do with it.

maximuswallis, 2025-02-25:
I had the same problem as this! Exactly the same. Rolled back and it is okay again.

Pho3niX90, 2025-02-26:
thanks for picking this up. will fix it now, might have been a merge conflict

0rangutan, 2025-02-26:
Time slots are back in 3.0.5 but now the switches to enable/disable them have disappeared for me.

Pho3niX90, 2025-02-26:
> switches to enable/disable the

There aren't switches to enable/disable them? Which ones are you referring to?

0rangutan, 2025-02-26:
All of the switches for charge/discharge disappeared, eg. switch.solis_grid_time_of_use_discharge_period_1

Pho3niX90, 2025-02-26:
> All of the switches for charge/discharge disappeared, eg. switch.solis_grid_time_of_use_discharge_period_1

can you just upload your logs please, as these haven't been changed in a while. That being said, make sure HA didn't create new inputs due to the new model name changes

charliegadget, 2025-02-26:
Confirmed the editable time windows are back and working.

Pho3niX90, 2025-02-26:
> Rolled back to 3.0.2 and all is okay. Forward to 3.0.5 again and all of the time of charge switches disappear again. Also, a number of other sensors are lost too, eg. sensor.solis_battery_discharge_power Log attached.
> 
> [home-assistant_solis_modbus_2025-02-26T11-23-20.773Z.log](https://github.com/user-attachments/files/18983110/home-assistant_solis_modbus_2025-02-26T11-23-20.773Z.log)

okay just create a separate issue for this one. there doesn't seem to be any errors in the logs.

0rangutan, 2025-02-26:
No, sorry!
Still missing a number of key entities each time I move beyond 3.0.2, eg.

[image]

Pho3niX90, 2025-02-26:
> No, sorry! Still missing a number of key entities each time I move beyond 3.0.2, eg.
> 
> [image]

can you just provide the logs for these as well?

charliegadget, 2025-02-26:
And here is is in 3.0.1

[image]

[image]

Pho3niX90, 2025-02-26:
> I have seen the same and can confirm 0rangutan's issue. Here is my log and screen shots.
> 
> [home-assistant_solis_modbus_2025-02-26T13-58-59.466Z.log](https://github.com/user-attachments/files/18985355/home-assistant_solis_modbus_2025-02-26T13-58-59.466Z.log)
> 
> <img alt="Image" width="343" src="https://private-user-images.githubusercontent.com/187340592/417173638-2b3f0386-fb9a-4bc5-8c48-a4e1090fcdbd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDA1OTM2NTMsIm5iZiI6MTc0MDU5MzM1MywicGF0aCI6Ii8xODczNDA1OTIvNDE3MTczNjM4LTJiM2YwMzg2LWZiOWEtNGJjNS04YzQ4LWE0ZTEwOTBmY2RiZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyNlQxODA5MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMzhjNDY2NWQ3ZDhhOTJhODZkODg1Y2ZmMDY2Y2M2Yjg3MTg4OTM0YjE4ZjE2OTViMGZhNTlmODEyMzAwNDcxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.CevjNNd8AbpC5YtHBVcQ5VJJKCY8Jh47xJ51vxp9Ke0"> <img alt="Image" width="592" src="https://private-user-images.githubusercontent.com/187340592/417173884-54650acb-b53f-4a98-bf89-0e535a3f4812.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDA1OTM2NTMsIm5iZiI6MTc0MDU5MzM1MywicGF0aCI6Ii8xODczNDA1OTIvNDE3MTczODg0LTU0NjUwYWNiLWI1M2YtNGE5OC1iZjg5LTBlNTM1YTNmNDgxMi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyNlQxODA5MTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYmNlY2U0MTQ2ZmNlNDlmNTRkNzQ3ODljNWYyMDNlNWRlNzFiNWY3NDJlMjdiNWEwMmE3MTBhMGYyOGM2MTEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.5TuFa8rrD3jLoiBy4ieo2LObEog0M-sF8h0ALBbew1c">

which inverter model did you choose?

0rangutan, 2025-02-26:
I have used Waveshare and S6-EH1P, same result.

Pho3niX90, 2025-02-26:
I am just trying to figure out the difference between setups. For me, those work, using a S6-EH1P as well

charliegadget, 2025-02-26:
I have always used this one and it works fine with 3.0.1. The actual inverter is S5-EH1P

0rangutan, 2025-02-26:
I also have the S5-EH1P inverter and mine works up to 3.0.2 then nothng since.

Pho3niX90, 2025-02-27:
@charliegadget @0rangutan 
Okay please create a separate issue for this, as to not lose the goal of this issue. 

I investigated the code base between 3.0.2 and latest, and there should be no reason that they do not provide the exact same sensors. I suspect that there might be a delay after upgrading. If you are convinced the modbus was connected, and data was being sent, then it needs to be investigated. To find the cause I would obviously need the debug logs, and the sensors that are disabled that are known to be working with your inverter. Changes in v3 include, that it would disable sensors now when the data received from the inverter are either problematic in length, or block response is too small. Prior to this, it would have always shown sensors enabled, even if they did not work. Lastly, do choose the correct model from the config, and restart HA to make sure that the config is correctly applied. These models help setup everything correctly for your inverter, and disables what is 100% known not to be working. I have added `S5-EH1P` to the dropdown in https://github.com/Pho3niX90/solis_modbus/releases/tag/3.0.7, for now it's identical to the S6, until we can filter out things known to not work

closed 2025-02-27 by Pho3niX90

charliegadget, 2025-02-27:
OK, not sure what you did, but all the switches are back and it all looks fine. When I updated to 3.0.7 an HA restart wasnt enough as the time issue I first reported on this thread returned, so I did a full reboot and everything is back as it should. I have selected the S5-EH1P option as well.

0rangutan, 2025-02-27:
Pleased to say that this approach seems to have resolved the issue for me too.
I hadn't considered a full reboot, but that sorted it.
Thanks!

charliegadget, 2025-03-07:
This issue has returned on 3.1.6 

[image]

[image]

commit 570dae1a (2025-02-26): fix: #150