Pho3niX90/solis_modbus #151: [Bug]: Multiple errors/warnings S 6 config on Solis RAI-3K-48ES-5G

Ian-Johnston, 2025-02-25:
## 🐛 Bug Description
Version 1.6.2
x86 box
HA = 2025.2.5
Using S6 Solis to connect to Solis RAI-3K-48ES-5G battery only inverter.

I got a collection of warnings/errors and am curious to what the mean. Whatever the issue is/was the system recovered at hasn't missed a beat.

Note:
Coincidentally, at 3am HA writes to a couple of the modbus registers to set Time of use charge start end slot #1.

Sorry, debug logging was not enabled.

Ian.

##########################################

Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:317
integration: Sensor (documentation, issues)
First occurred: 03:00:10 (9 occurrences)
Last logged: 12:20:10

Entity sensor.solis_s6_solis_today_energy_imported_from_grid from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 27.3 (28.2) with last_updated set to 2025-02-25T06:40:40.219786+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
Entity sensor.total_energy_discharged from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 3382.0 (3383.0) with last_updated set to 2025-02-25T06:43:26.076923+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.monthly_energy_discharged from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 250.0 (251.0) with last_updated set to 2025-02-25T06:43:26.076654+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.yearly_energy_discharged from integration solis has state class total_increasing, but its state is not strictly increasing. Triggered by state 558.0 (559.0) with last_updated set to 2025-02-25T06:43:26.076792+00:00. Please create a bug report at https://github.com/hultenvp/solis-sensor/issues
Entity sensor.solis_s6_solis_pv_total_energy_generation from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 3171 (3172.0) with last_updated set to 2025-02-25T12:19:11.642828+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

##########################################

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:52
integration: Solis Modbus (documentation, issues)
First occurred: 00:16:50 (4 occurrences)
Last logged: 12:18:54

No response received after 3 retries, continue with next request

##########################################

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:63
integration: Solis Modbus (documentation, issues)
First occurred: 02:58:12 (1 occurrences)
Last logged: 02:58:12

No response received after 3 retries, continue with next request

##########################################

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:84
integration: Solis Modbus (documentation, issues)
First occurred: 00:03:20 (1 occurrences)
Last logged: 00:03:20

No response received after 3 retries, continue with next request

##########################################

## 🔄 To Reproduce

## ✅ Expected Behavior

## 📸 Screenshots / Logs

## 🔧 System Information
Solis_Modbus = Version 1.6.2
x86 box
HA = 2025.2.5

### 🔹 Integration:
Dual Waveshare, hard wired.

## ➕ Additional Context

Ian-Johnston, 2025-02-26:
>irst occurred: 03:00:10 (9 occurrences)
>Last logged: 12:20:10

The first occurence, yes, it is immediately after a write to the inverter.......but the last logged is not.

I think I spotted an issue in the logs above, that first report mentions **pv**, and of course my inverter is battery only, no PV.
Also, the report mentions not only your repository but https://github.com/**hultenvp**/solis-sensor/issues also.
I'll try and remove any PV from said **hultenvp** code.

Ian.

Pho3niX90, 2025-02-26:
Which version of the integration are you using? And just to be sure, if it's 3+, which model did you choose? 

Will apply a bugfix today

closed 2025-03-07 by Pho3niX90