Pho3niX90/solis_modbus #152: [Bug]:

cemasss, 2025-02-25:
## 🐛 Bug Description
When connected Solis modbus to Home Assistant, inverter goes offline on Solis cloud.

## 🔄 To Reproduce
Steps to reproduce the behavior:
1. when I turn on modbus, after 10-20 minutes the inverter disconnects from solis cloud

## ✅ Expected Behavior
it would be great if both modbus and solis cloud worked at the same time

🔧 System Information
🔹 Home Assistant:
- Version:  core-2025.2.4
- Installation Method**:  Home Assistant green

### 🔹 Integration:
- **Plugin Version**: [3.03]
- **Device Model**: [e.g., Solis S6-EH3P15K-H]
- **Modbus Connection Type?**: [Wifi]
- **Custom Config Entries?**: [No]

hbpv, 2025-02-25:
Hi

This is normal - you can only be connected to one of both: Modbus OR Soliscloud.

Solis itself declares that in a support article.

Best
Max

0rangutan, 2025-02-26:
As Max says, this is normal behaviour and not something that can be changed in this integration - the Solis Data Logger will only accept one TCP/IP connection.

To work around this, if you want to use both this integration and Solis Cloud, look at hardware setups like this one: https://github.com/alienatedsec/solis-ha-modbus-cloud

It's a little effort and expense, but works well.

closed 2025-02-26 by Pho3niX90

koen-db, 2025-07-21:
I have this inverter: S6-EH3P8K02-NV-YD-L and the WL datalogger.
Connected over WiFi or LAN I have had both Modbus over TCP _and_ Soliscloud access for the past weeks.

Specific firmware version for the datalogger: 100121b4
Maybe they changed something recently?