Pho3niX90/solis_modbus #155: [Enhancement]: S6-EH3010K-H-ZP

Dominic-070, 2025-02-28:
Hi the switch entities disappeared in one of the latest updates for Wavshare. This whaveshare option is used in the Netherlands for the Zonneplan Nexus inverters model S6-EH3P10K-H-ZP. because it is supplied without a data logging connection. This has the same functions as the S6-EH3P, but no connection for solar panels and the multipliers are therefore different. Is it an idea to rename Wavshare to S6-EH3P10K-H-ZP and even give them the same specifications as the S6-EH3P only with the different multipliers? And restore the switch entities ?

[image]

[image]

0rangutan, 2025-02-28:
Try a full reboot of your HA server, rather than reatarting the HA service.
This fixed missing switches for a few users on a recent release.

Dominic-070, 2025-02-28:
unfortunately that doesn't work either. I have to go back to 3.0.5 to see the switches again and they still don't work.

[image]

Dominic-070, 2025-03-01:
Removed from integration updated to 3.1.0 add with S6-EH3P. Works perfectly with S6-EH3P (apart from the multipliers).

[image]

[image]

Removed again, full reboot and added with waveshare. Again full reboot, Missing al the switch entities., all the time entities are not working, 

[image]

[image]

Dominic-070, 2025-03-01:
I enabled debug logging and restarted the integration 2 times:
`
2025-03-01 12:45:42.144 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 33243 starting
2025-03-01 12:45:42.176 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Register 33243: [0, 0, 0]
2025-03-01 12:45:42.176 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] inter_frame elapsed = 3.0768562150001344, remaining_wait = 0.0
2025-03-01 12:45:42.176 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33243, register 33243 has value 0
2025-03-01 12:45:42.180 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33243, register 33244 has value 0
2025-03-01 12:45:42.184 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33243, register 33245 has value 0
2025-03-01 12:45:42.188 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 43010 starting
2025-03-01 12:45:42.312 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Holding Register 43010: [100, 7, 500, 500, 6100, 1000, 6000, 6000, 4, 0, 0, 0, 0, 0, 80, 0, 0, 30, 0]
2025-03-01 12:45:42.312 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] inter_frame elapsed = 0.13577103299985538, remaining_wait = 0.1642289670001446
2025-03-01 12:45:42.479 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43010 has value 100
2025-03-01 12:45:42.483 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43011 has value 7
2025-03-01 12:45:42.487 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43012 has value 500
2025-03-01 12:45:42.490 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43013 has value 500
2025-03-01 12:45:42.498 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43014 has value 6100
2025-03-01 12:45:42.502 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43015 has value 1000
2025-03-01 12:45:42.509 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43016 has value 6000
2025-03-01 12:45:42.513 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43017 has value 6000
2025-03-01 12:45:42.518 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43018 has value 4
2025-03-01 12:45:42.522 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43019 has value 0
2025-03-01 12:45:42.526 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43020 has value 0
2025-03-01 12:45:42.532 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43021 has value 0
2025-03-01 12:45:42.539 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43022 has value 0
2025-03-01 12:45:42.544 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43023 has value 0
2025-03-01 12:45:42.550 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43024 has value 80
2025-03-01 12:45:42.556 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43025 has value 0
2025-03-01 12:45:42.560 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43026 has value 0
2025-03-01 12:45:42.566 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43027 has value 30
2025-03-01 12:45:42.570 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43010, register 43028 has value 0
2025-03-01 12:45:42.574 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ✅ NORMAL update completed in 0.4298s
2025-03-01 12:45:43.232 DEBUG (MainThread) [custom_components.solis_modbus] init async_unload_entry
2025-03-01 12:45:43.250 DEBUG (MainThread) [custom_components.solis_modbus] {'host': '192.168.1.50', 'port': 502, 'poll_interval_fast': 10, 'poll_interval_normal': 15, 'poll_interval_slow': 30, 'model': 'WAVESHARE', 'has_v2': False, 'has_pv': False, 'has_battery': True, 'has_generator': False}
2025-03-01 12:45:43.250 INFO (MainThread) [custom_components.solis_modbus] Loaded Solis Modbus Integration with Model: WAVESHARE
2025-03-01 12:45:43.250 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 33000, sensor count = 5, registrar count = 20
2025-03-01 12:45:43.251 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 35000, sensor count = 1, registrar count = 1
2025-03-01 12:45:43.251 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 33025, sensor count = 3, registrar count = 3
2025-03-01 12:45:43.251 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 33029, sensor count = 7, registrar count = 12
2025-03-01 12:45:43.251 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 33049, sensor count = 9, registrar count = 10
[... cut ...]
2025-03-01 12:45:55.984 INFO (MainThread) [custom_components.solis_modbus.number] Solis platform_config: {'host': '192.168.1.50', 'port': 502, 'poll_interval_fast': 10, 'poll_interval_normal': 15, 'poll_interval_slow': 30, 'model': 'WAVESHARE', 'has_v2': False, 'has_pv': False, 'has_battery': True, 'has_generator': False}
2025-03-01 12:45:55.984 INFO (MainThread) [custom_components.solis_modbus.number] Number entities = 55
2025-03-01 12:45:56.019 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Connecting to Modbus TCP...
2025-03-01 12:45:56.123 INFO (MainThread) [custom_components.solis_modbus.data_retrieval] ✅ Modbus controller connected successfully.

`

Dominic-070, 2025-03-01:
Also removed the inverter from the integration, enabled debug logging en added the inverter again.
see log file (to big to paste here):
[log solis.txt](https://github.com/user-attachments/files/19038762/log.solis.txt)

Pho3niX90, 2025-03-01:
please try https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.2

select the appropriate inverter, then select waveshare as your connection type

Dominic-070, 2025-03-01:
Hi I have completely removed the integration and reinstalled the new version. Also did a complete restart of HA. However, the switch entities remain unusable except for 1.

[image]

```
2025-03-01 18:34:38.222 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Register 33251: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2025-03-01 18:34:38.222 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] inter_frame elapsed = 0.10418904099992687, remaining_wait = 0.19581095900007311
2025-03-01 18:34:38.419 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33251 has value 0
2025-03-01 18:34:38.422 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33252 has value 0
2025-03-01 18:34:38.423 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33253 has value 0
2025-03-01 18:34:38.424 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33254 has value 0
2025-03-01 18:34:38.424 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33255 has value 0
2025-03-01 18:34:38.425 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33256 has value 0
2025-03-01 18:34:38.425 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33257 has value 0
2025-03-01 18:34:38.425 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33257: 0}
2025-03-01 18:34:38.425 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33258 has value 0
2025-03-01 18:34:38.426 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33259 has value 0
2025-03-01 18:34:38.426 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33259: 0}
2025-03-01 18:34:38.426 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33260 has value 0
2025-03-01 18:34:38.427 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33261 has value 0
2025-03-01 18:34:38.427 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33261: 0}
2025-03-01 18:34:38.427 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33262 has value 0
2025-03-01 18:34:38.427 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33263 has value 0
2025-03-01 18:34:38.427 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33263: 0}
2025-03-01 18:34:38.428 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33251, register 33264 has value 0
2025-03-01 18:34:38.428 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 33530 starting
2025-03-01 18:34:38.504 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Register 33530: [0, 0, 0, 0, 0, 0]
2025-03-01 18:34:38.505 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] inter_frame elapsed = 0.08605956999997488, remaining_wait = 0.2139404300000251
2025-03-01 18:34:38.720 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33530 has value 0
2025-03-01 18:34:38.723 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33531 has value 0
2025-03-01 18:34:38.725 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33532 has value 0
2025-03-01 18:34:38.726 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33532: 0}
2025-03-01 18:34:38.728 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33533 has value 0
2025-03-01 18:34:38.731 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33534 has value 0
2025-03-01 18:34:38.733 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33535 has value 0
2025-03-01 18:34:38.736 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ✅ FAST update completed in 1.8679s
2025-03-01 18:34:41.643 DEBUG (MainThread) [custom_components.solis_modbus] init async_unload_entry
2025-03-01 18:34:41.717 DEBUG (MainThread) [custom_components.solis_modbus] {'host': '192.168.1.50', 'port': 502, 'poll_interval_fast': 10, 'poll_interval_normal': 15, 'poll_interval_slow': 30, 'model': 'S6-EH3P', 'connection': 'WAVESHARE', 'has_v2': False, 'has_pv': False, 'has_battery': True, 'has_generator': False}
2025-03-01 18:34:41.717 INFO (MainThread) [custom_components.solis_modbus] Loaded Solis Modbus Integration with Model: S6-EH3P
2025-03-01 18:34:41.717 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 33000, sensor count = 5, registrar count = 20
2025-03-01 18:34:41.717 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 35000, sensor count = 1, registrar count = 1
2025-03-01 18:34:41.717 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_base_sensor] Sensor group creation. start registrar = 33025, sensor count = 3, registrar count = 3
[... cut ...]
2025-03-01 18:35:03.847 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33533 has value 0
2025-03-01 18:35:03.847 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33534 has value 0
2025-03-01 18:35:03.848 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33530, register 33535 has value 0
2025-03-01 18:35:03.849 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ✅ FAST update completed in 1.9669s

```

Dominic-070, 2025-03-01:
It works when i check the V2 Firmware

[image]

[image]

Pho3niX90, 2025-03-01:
Perfect. Just keep in mind, that most setting changes would require a restart of the integration. 

V2 Firmware only changes the grid switches, and the grid time info. Keep this on, as this is applicable for all new firmwares

Modbus can take a while before fully utilized, a minimum of your slowest setting. So, give it about 10mins to settle and read all info from the inverter first, including settings.

Dominic-070, 2025-03-01:
Thnx!

closed 2025-03-01 by Dominic-070