Pho3niX90/solis_modbus #157: [Help]: How to set Modbus andress

rocktester, 2025-03-01:
## ❓ Configuration Help Request
Hallo ...

i am using waveshare TCP server on LAN for Modbus connection to different devices on different Modbus adresses).
It works ok with Node RED  with Tem/HUm sensor Adr1 and Solis Inverter Adr2.

Question: How to set the Modbus adress of the Inverter in this integration ?

Regards Mike

Pho3niX90, 2025-03-01:
I am not understanding the question. What do you mean? Can you show me an example.

rocktester, 2025-03-01:
Hallo .....
modbus can support upto 255 devices on the 2 wire bus.
The parameters device can be selected with the Modbus adress.
like here in the SOLAX integration (but it als not working in HomeAssistant)

[image]

In Node red you can select also the adress of the device on the bus.
exmaples:
Ii use this connection for the Temp/Hum sensor (SHT20) on Modbus Adress 1 via the Waveshare GW on the LAN

[image]

and this is for read the values of solis inverter on Modbus Adress 2 via the Waveshare gateway on the LAN

[image]

i hope this helps to understand the problem.
Regards Mike

Pho3niX90, 2025-03-01:
Got you, current you cannot change the slave for this integration. Will add it in the next version today/tomorrow

rocktester, 2025-03-01:
very NICE .... THANK YOU VERY MUCH

Pho3niX90, 2025-03-02:
Implemented in https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.3

closed 2025-03-02 by Pho3niX90

rocktester, 2025-03-03:
nice..... it works 

[image]

THANK YOU VERY MUCH

commit 0f0dea71 (2025-03-02): feat: ability to change slave address, fixes #157