Pho3niX90/solis_modbus #158: [Bug]: RC Force change/discharge conflict

wawyed, 2025-03-03:
## 🐛 Bug Description
@Pho3niX90 I think the switches conflict with each other. I turned discharge on and after 5 minutes it went off on its own.

## 🔄 To Reproduce
Steps to reproduce the behavior:

Turn RC force discharge on.

Wait 5 minutes

RC force discharge turns off on its own.

## ✅ Expected Behavior
RC force discharge stays on until switched back off. Maybe switching to a dropdown would fix the problem?

## 📸 Screenshots / Logs

[image]

wawyed, 2025-03-03:
If it's not that, the only other thing I could think of is solis cloud turning it off ?

wawyed, 2025-03-03:
I got some logs:

```
12.225 DEBUG (SyncWorker_0) [custom_components.solis_modbus.sensors.solis_binary_sensor] 43135-None turn on called 
	Line 16292: 2025-03-03 18:53:12.226 DEBUG (SyncWorker_0) [custom_components.solis_modbus.sensors.solis_binary_sensor] Attempting bit None to True in register 43135. New value for register 2
	Line 16293: 2025-03-03 18:53:12.226 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Queued Write Holding Register register = 43135, value = 2
	Line 16295: 2025-03-03 18:53:12.334 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Write Holding Register register = 43135, value = 2, int_value = 2: WriteSingleRegisterResponse(dev_id=1, transaction_id=61611, address=43135, count=0, bits=[], registers=[2], status=1)
	Line 16296: 2025-03-03 18:53:12.336 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 16297: 2025-03-03 18:53:12.336 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 16298: 2025-03-03 18:53:12.336 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 2, is_on = True, 
	Line 16299: 2025-03-03 18:53:12.336 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_2 set to True
	Line 16300: 2025-03-03 18:53:12.336 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 16301: 2025-03-03 18:53:12.337 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 16302: 2025-03-03 18:53:12.337 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 2, is_on = True, 
	Line 16303: 2025-03-03 18:53:12.337 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_2 set to True
	Line 16304: 2025-03-03 18:53:12.337 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 16305: 2025-03-03 18:53:12.337 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 16395: 2025-03-03 18:53:14.648 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43129, register 43135 has value 2
	Line 16396: 2025-03-03 18:53:14.651 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 16397: 2025-03-03 18:53:14.651 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 16398: 2025-03-03 18:53:14.651 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 2, is_on = True, 
	Line 16399: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_2 set to True
	Line 16400: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 16401: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 16402: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 2, is_on = True, 
	Line 16403: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_2 set to True
	Line 16404: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 16405: 2025-03-03 18:53:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 17284: 2025-03-03 18:53:44.645 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43129, register 43135 has value 2
	Line 17285: 2025-03-03 18:53:44.648 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 1, is_on = False, 
	Line 17286: 2025-03-03 18:53:44.648 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False
	Line 17287: 2025-03-03 18:53:44.649 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 2, on_value = 2, is_on = True, 
[... cut ...]
	Line 25293: 2025-03-03 18:58:14.651 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_2 set to False
	Line 25294: 2025-03-03 18:58:14.651 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] Sensor update received, register = 43135, value = 0, on_value = 1, is_on = False, 
	Line 25295: 2025-03-03 18:58:14.652 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_binary_sensor] switch solis_modbus_192.168.50.65_43135_1 set to False

wawyed, 2025-03-03:
I also found this in the documentation:

[image]

It's for a different registry but I wonder if it expects the same from the other one? It seems to match the 5 minutes delay and off

wawyed, 2025-03-04:
I did some more research and I found someone with the same problem:

https://diysolarforum.com/threads/communication-to-solis-rai-3k-48es-5g-energy-storage-inverters-using-modbus.95482/page-2#post-1284998

> It is, however it's not the datalogger - it's the inverter doing it. Setting that register to 1 (or presumably 2 for discharge), seems to set a 5-minute timer which then clears the register again. This happens even with the datalogger removed, and it's always 5 minutes after the register was set regardless of the datalogger's last poll.
> 
> My guess is this is so Solis can trigger a charge/discharge remotely and it will automatically turn off unless they reset before the timer expires. That would prevent the charge period going on too long if the datalogger polling fails.
> 
> I think I'm just going to have to poke that register every 5 minutes while I'm using it to charge.
> 
> I'll test that tonight/tomorrow when the Agile prices drop right down again.

So I don't think it's a problem of the integration but rather a "feature" of the inverter.

For now I'm writting to the registry using your service modbus write holding registry before the 5 minutes is past to reset the timer.

Pho3niX90, 2025-03-04:
Interesting. Not entirely sure how to handle this.

I would probably have to add an input for the timer, and trigger it every 5minutes, and disable it at the end.

wawyed, 2025-03-04:
Fow now, I have created an automation as such, not sure if it helps

``` yaml
alias: Keep Remote Force Battery Discharge running
description: ""
triggers:
  - trigger: state
    entity_id:
      - switch.solis_s6_solis_rc_force_battery_discharge
    from: "off"
    to: "on"
conditions: []
actions:
  - delay:
      hours: 0
      minutes: 4
      seconds: 50
      milliseconds: 0
  - repeat:
      sequence:
        - action: solis_modbus.solis_write_holding_register
          data:
            address: 43135
            value: 2
        - delay:
            hours: 0
            minutes: 4
            seconds: 50
            milliseconds: 0
      while:
        - condition: state
          entity_id: switch.solis_s6_solis_rc_force_battery_discharge
          state: "on"
mode: single

hbpv, 2025-03-05:
Related to #123 ?
Maybe thinking about an dropdown again? 💃

wawyed, 2025-03-05:
It is not related to having two switches rather it is a "feature" of the inverter. I have a ticket open with Solis to see what they say.

wawyed, 2025-03-06:
I got this from solis (which I asked them to clarify), not sure if it makes any sense to you @Pho3niX90 

[image]

Pho3niX90, 2025-03-06:
Function code 10, is just a method to write multiple registers at the same time. Meaning, they are hinging there might be a register to set the timeout. 

Looking at other registers with the same timeout:

[image]

Suggests this one
[image]

Pho3niX90, 2025-03-06:
can you perhaps issue a 3 minute to 43282, and see if it changes? If so, this would be an easy change

wawyed, 2025-03-06:
hmm interesting idea, let me try

wawyed, 2025-03-06:
I just checked the documentation and that registry doesn't show in my API doc

wawyed, 2025-03-06:
Although the doc says 30min max value which is not great...

Pho3niX90, 2025-03-06:
> Although the doc says 30min max value which is not great...

It's a U16 registry, so longer "should" work, if it actually writes. Will implement this registry

Pho3niX90, 2025-03-07:
added in https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.5

closed 2025-03-07 by Pho3niX90