Pho3niX90/solis_modbus #159: Tonnes of errors in logs

miketarg, 2025-03-04:
Logger: pymodbus.logging
Source: runner.py:154
First occurred: 3:48:49 AM (3766 occurrences)
Last logged: 2:16:25 PM

Exception response 132 / 0

Solis 
3.1.3

Core
2025.3.0b1

Supervisor
2025.03.0

Operating System
14.2

Frontend
20250227.0

veletron2, 2025-03-04:
Yes, I have same, below is after a restart, but there were thousands beforehand. 

Logger: pymodbus.logging
Source: runner.py:154
First occurred: 17:45:29 (13 occurrences)
Last logged: 17:47:29

Exception response 132 / 0
**Unable to decode frame Modbus Error: Unknown response 0**

veletron2, 2025-03-04:
Looks like an error decoding Group 34328 ? here is a debug excerpt from the log. The error is repeated always after the same group.

`2025-03-04 17:51:55.249 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33149: 0}
2025-03-04 17:51:55.249 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_derived_sensor] not all values received yet = {33135: 1, 33149: 0}
2025-03-04 17:51:55.249 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_derived_sensor] not all values received yet = {33135: 1, 33149: 0}
2025-03-04 17:51:55.250 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33132, register 33150 has value 2116
2025-03-04 17:51:55.250 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33132, register 33151 has value 0
2025-03-04 17:51:55.251 DEBUG (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] not all values received yet = {33151: 0}
2025-03-04 17:51:55.251 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 33132, register 33152 has value 1876
2025-03-04 17:51:55.252 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 34328 starting
2025-03-04 17:51:55.480 ERROR (MainThread) [pymodbus.logging] Exception response 132 / 0
2025-03-04 17:51:55.480 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Register 34328: []
2025-03-04 17:51:55.480 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] inter_frame elapsed = 0.24292266499924153, remaining_wait = 0.057077335000758456
2025-03-04 17:51:55.538 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️ Modbus read mismatch: Received 0 values, expected 6 for register 34328 - 34333. Skipping because linking them is uncertain.
2025-03-04 17:51:55.539 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 33251 starting`

hbpv, 2025-03-05:
Would be great for @Pho3niX90 if you could also provide him the inverter model (string/hybrid/name)

Pho3niX90, 2025-03-05:
34328 is related to the smart port. Cannot find information which inverters support it exactly. So will limit it to the S6 models.

Pho3niX90, 2025-03-05:
See if this helps @veletron2 https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.4

veletron2, 2025-03-05:
So, that error appears to have gone, replaced with a warning thus. I assume this is expected given the changes?

Logger: custom_components.solis_modbus
Source: custom_components/solis_modbus/__init__.py:172
integration: Solis Modbus (documentation, issues)
First occurred: 16:14:16 (1 occurrences)
Last logged: 16:14:16

Skipping sensor group 'Unnamed' due to missing required features: [<InverterFeature.SMART_PORT: ('smart_port',)>]

Pho3niX90, 2025-03-06:
yes thats just a debug log, stating why some sensors aren't available. You can turn off debug mode.