Pho3niX90/solis_modbus #162: [Feature]: Add Max Charge/Discharge Battery current inputs

wawyed, 2025-03-05:
## 🚀 Feature Request
I'd like to be able to control the max charge/discharge current (A)

## 🔍 Use Case
It is useful when you don't want the battery to ever go above a certain current

## ✅ Expected Behavior
As with time charge/current current a way to set the overall max charge/discharge

## 🔗 Additional Context

Max is 100A in my case

[image]

Pho3niX90, 2025-03-06:
Registrars starting with `3` arre only readable, `4` settable. 

I believe the ones would then be 43012 and 43013

[image]

wawyed, 2025-03-06:
hmm you are right my bad. I have this ones for my inverter at least:

[image]

 The ones you mention say not used for mine:

[image]

Pho3niX90, 2025-03-07:
added in https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.5

closed 2025-03-07 by Pho3niX90