Pho3niX90/solis_modbus #163: [Bug]: reset config after reboot

Dominic-070, 2025-03-06:
When i reboot HA, the congig goes back to default. It seems to forget to remeber conection type. Multiepliers are reset after reboot.

[image]

Pho3niX90, 2025-03-07:
are you seeing the config reset by means of data received and debug logs. Or does it appear to reset due to the config UI?

It might be that the config relay is buggy, or the actual config is buggy.

Dominic-070, 2025-03-07:
I see after a restart of home assistant that the values ​​have changed. The adjusted multipliers of waveshare are no longer applied. 
Also reset the set values ​​in UI

Orginal settings:

[image]

After restart HA

[image]

Pho3niX90, 2025-03-07:
try https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.6

closed 2025-03-07 by Pho3niX90

Dominic-070, 2025-03-07:
Yes perfect!!!!
thnx!!!!