Pho3niX90/solis_modbus #164: [Bug]: S5 Time windows are unedittable again

charliegadget, 2025-03-07:
## 🐛 Bug Description
See my previous bug report that is not closed. The issue has returned in 3.1.6 everything is working in 3.1.4

https://github.com/Pho3niX90/solis_modbus/issues/150

[image]
[image]

0rangutan, 2025-03-07:
Confirmed, same issue here

Arnoudd, 2025-03-07:
Same issue here. I reverted to 3.1.3 and it works again.

Pho3niX90, 2025-03-07:
There were some time changes in todays release, will fix this shortly

closed 2025-03-07 by Pho3niX90

charliegadget, 2025-03-07:
Thanks. Confirmed as fixed for me.

commit bb9025d9 (2025-03-07): fixes #164