Pho3niX90/solis_modbus #165: [Bug]: Accumulative entities must be kWh and not kW

wgentine, 2025-03-11:
The entities below must be kWh rather kW since kW is used for momentary and kWh for accumulative:

[image]

The same goes for this one:

[image]

The second is likely correct but isnt reporting any data.

This has a negative effect on home assistant energy dashboard for example. (device class for these also must be of power and state total_increasing)

Pho3niX90, 2025-03-11:
> For S6-EH3P the values seem to be legit: (@Pho3niX90 please be careful :)
> 
> [image]
> 

The GR1P uses completely different sensors than the rest, so no worries.

Pho3niX90, 2025-03-12:
Let me know if this solves the issue
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.8

wgentine, 2025-03-12:
Thank you @Pho3niX90 ! Now the unit is correct.

However, something in their configurations still seems to be off since I cannot select any of them for the Energy Dashboard:

[image]

wgentine, 2025-03-12:
Ahh, I see... "device_class": SensorDeviceClass.POWER needs to be ENERGY instead.

Pho3niX90, 2025-03-13:
Should have seen that, here is the fix https://github.com/Pho3niX90/solis_modbus/releases/tag/3.1.9

wgentine, 2025-03-13:
Thanks @Pho3niX90 ! All perfect now. Thank you!

commit c0448dd9 (2025-03-12): fixes #165
commit 860c3a52 (2025-03-12): fixes #165
commit 759bef7f (2025-03-12): fixes #165
closed 2025-03-13 by wgentine