Pho3niX90/solis_modbus #166: [Bug]: Battery Charge/Discharge Limit Power

larsfrisk, 2025-03-12:
## 🐛 Bug Description
A clear and concise description of what the bug is.

The Charge and Discharge limits are hard in code set to maximum 6000W.  My S6-EH3P12k-H and BYD-HVS 12,6 can do 12kW, and I need to be able to set the limitation that high in order to get full payment for my grid support. services  

## 🔄 To Reproduce
Steps to reproduce the behavior:
1. **Go to**: [e.g., Energy Dashboard, Developer Tools → States]
2. **Click on**: [e.g., a specific sensor, switch, or setting]
3. **Perform action**: [e.g., toggling a switch, checking logs]
4. **Expected result**: [e.g., switch should turn on, value should update]
5. **Actual result**: [e.g., state doesn't change, wrong reading]

Set Solis S6-EH3P Battery Charge Limit Power to a value over 6000. Response is that 6000 is maximum.
I modified manually in hybrid_sensors.py and then I can set up to 12000

## ✅ Expected Behavior
A clear and concise description of what you expected to happen.

## 📸 Screenshots / Logs
If applicable, add screenshots or logs to help explain your problem.
- **Home Assistant logs** (filtered to relevant errors)

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: [e.g., core-2024.2.1]
- **Installation Method**: [e.g., Home Assistant OS, Docker, Supervised]

### 🔹 Integration:
- **Plugin Version**: [e.g., 1.2.3]
- **Device Model**: [e.g., Solis 5G 6kW, Axitec 10kW]
- **Modbus Connection Type?**: [e.g., Wifi, Ethernet, Waveshare]
- **Custom Config Entries?**: [Yes/No]

## ➕ Additional Context
Any other details that might be helpful for debugging.

hbpv, 2025-03-12:
@Pho3niX90 already mentioned this for Axitec wich is the Same Solis:

https://github.com/Pho3niX90/solis_modbus/issues/123#issuecomment-2690500910

larsfrisk, 2025-03-12:
> [@Pho3niX90](https://github.com/Pho3niX90) already mentioned this for Axitec wich is the Same Solis:
> 
> [#123 (comment)](https://github.com/Pho3niX90/solis_modbus/issues/123#issuecomment-2690500910)

Right. But that is closed?

hbpv, 2025-03-12:
> > [@Pho3niX90](https://github.com/Pho3niX90) already mentioned this for Axitec wich is the Same Solis:
> > [#123 (comment)](https://github.com/Pho3niX90/solis_modbus/issues/123#issuecomment-2690500910)
> 
> Right. But that is closed?

Maybe just lost in conversation… and Not that easy because of different models.. lets wait for phoenix

Pho3niX90, 2025-03-14:
> [@Pho3niX90](https://github.com/Pho3niX90) already mentioned this for Axitec wich is the Same Solis:
> 
> [#123 (comment)](https://github.com/Pho3niX90/solis_modbus/issues/123#issuecomment-2690500910)

Yes that was related to RC, so when the issue at hand is resolved, the item gets closed. Otherwise requirements per issue just stacks.

Pho3niX90, 2025-03-14:
I will implement a input on the config to set the inverters maxes. Currently even going by model makes it a guessing game, and the last thing I want is for someones inverter to blow up due to maxes set to high from my end, although there should be safeguards on the inverters end, one can never be too careful.

commit 9c78c5db (2025-03-31): fixes: #166
fixes: #173
closed 2025-03-31 by Pho3niX90