Pho3niX90/solis_modbus #168: [Bug]: Unit coefficient

Raphael-zoreille, 2025-03-16:
Hello,

I've set up Solis Modbus, and there are coefficient mismatches on certain values:

[image]

[image]

[image]

------
On Solis cloud, the energy values ​​go back to 0.01 but in modbus they are in 1:

[image]

[image]

Thank you for the work done, it is very useful.

### 🔹 Integration:
- **Plugin Version**: 3.19
- **Device Model**: S5-EH1P6K-L
- **Modbus Connection Type?**: Waveshare

Pho3niX90, 2025-03-16:
If you enbaled the waveshare option, try disabling it.

Raphael-zoreille, 2025-03-17:
Hello,

As indicated in my post, the coefficients are incorrect.
For the battery kWh, you need to do a factor of 100.
For the BMS, you need a factor of 10.

Sincerely,

Pho3niX90, 2025-03-18:
@wawyed @miketarg @richvs @0rangutan @ctepelea
Could you please confirm if you are seeing the same issue on your side?

0rangutan, 2025-03-18:
I don't fully understand the issue being experienced.
My "Solis Total Battery Charge Energy" is 4,828 kWh, while in the Solis portal it is 4.828MWh - looks the same to me.

Pho3niX90, 2025-03-18:
@Raphael-zoreille did you enable waveshare in the config? As this changes the multiplier

wawyed, 2025-03-18:
This is what I'm getting 

Solis_cloud:

[image]

Solis_Modbus:

[image]

Not sure if there's some float overflow on the discharge? As the number looks off

Pho3niX90, 2025-03-18:
Docs have been updated to explain the waveshare enablement: https://solis-modbus.readthedocs.io/en/latest/sensors.html#waveshare

Raphael-zoreille, 2025-03-19:
Hello,

I disabled waveshare mode, here is what I get:

[image]

[image]

If I put the mode back here is what it gives:

[image]

[image]

[image]

The cloud indicates this:

[image]

🙂

Pho3niX90, 2025-03-20:
> Hello,
> 
> I disabled waveshare mode, here is what I get:
> 
> [image]
> 
> [image]
> 
> If I put the mode back here is what it gives:
> 
> [image]
> 
> [image]
> 
> [image]
> 
> The cloud indicates this:
> 
> [image]
> I have the impression that it is not the coefficient that is modified but only the display in 0.01
> 
> 🙂

Keep in mind that changing the waveshare option won't change historical values. 

Between the switches, dod you give it an hour for collection?

Raphael-zoreille, 2025-03-23:
Indeed, I hadn't waited long enough.
The values ​​are correct.

Thanks for the help.
And thanks for developing the plugin.👌

closed 2025-03-23 by Raphael-zoreille