Pho3niX90/solis_modbus #170: [Bug]: Solis S6GR1P1.5K fails to reconnect in morning

funks81, 2025-03-20:
Many thanks for all your work on this integration.  I have recently used it to connect my Solis S6GR1P1.5K string inverter (no batteries).  Previously I was using SolisCloud to connect (but obviously only had 5 minute updates).  The integration is working fine during the day - but at night the inverter switches off - and the integration continues to report a 20W solar output (minimum amount that is reported by this particular inverter).  In the morning - the inverter fails to reconnect - and continues to display the previous days values until I restart the Elfin  EW11 (RS485 to wifi) adaptor that I use to connect the inverter to home assistant.

I believe other integrations - including SolisCloud have previously had similar issues - and at first used some sort of midnight resetting to adjust this - and then i think they wrote something where a continuous 20w value was ignored.  To be honest - I'm not so bothered by the constant 20w value being reported - its that I have to manually restart the RS485 adaptor each morning - and this only works - once the inverter has switched on - and this time obviously changes from day to day.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:135
integration: Solis Modbus ([documentation](https://solis-modbus.readthedocs.io/), [issues](https://github.com/Pho3niX90/solis_modbus/issues))
First occurred: 18 March 2025 at 20:33:54 (1963 occurrences)
Last logged: 06:50:06

Failed to read input register 3033: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.68.129:502]
Failed to read input register 3250: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.68.129:502]
Failed to read input register 36028: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.68.129:502]
Failed to read input register 2999: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
Failed to read input register 3179: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request

Pho3niX90, 2025-03-20:
What rs485 device do you need to restart?

funks81, 2025-03-20:
> Pho3niX90
>  left a comment
> (Pho3niX90/solis_modbus#170)
> What rs485 device do you need to restart?

The Elfin EW11 RS485 to wifi adaptor - that is connected directly to the inverter.  I’ve only had the integration running for 2 days - so sure there may be other ways to get data again each morning.

Thanks

funks81, 2025-03-20:
I know nothing about coding - and only ever follow simple detailed instructions - but think this is the issue - and how it was fixed on the soliscloud integration

[https://github.com/hultenvp/solis-sensor/pull/355](url)

`        for attribute in data.keys():
            if attribute in self._subscriptions[serial]:
                value = getattr(data, attribute)
                if attribute == INVERTER_ENERGY_TODAY or attribute == INVERTER_ACPOWER:
                    # Energy_today and AC power are not reset at midnight, but in the 

                if attribute == INVERTER_ACPOWER and getattr(data, INVERTER_STATE) == 2:
                    # Overriding stale AC Power value when inverter is offline
                    value = 0
                elif attribute == INVERTER_ENERGY_TODAY:
                    # Energy_today is not reset at midnight, but in the 
                    # morning at sunrise when the inverter switches back on. This 
                    # messes up the energy dashboard. Return 0 while inverter is 
                    # still off.`

Pho3niX90, 2025-03-21:
I am confused. Is the issue that the connection doesn't reestablish. Or that the reset doesn't happen at the right time? 

The rest is an easy solution.

funks81, 2025-03-21:
The connection doeesn't re-establish without manually power cycling the Elfin EW11 (RS485 to wifi) that I use for Modbus to HA.

funks81, 2025-03-26:
So - i've got the EW11 to reconnect in the morning - by powercycling from sunrise until i get an updated power figure from the inverter - obviously not ideal - but an automated workaround.

commit bb7011fc (2025-04-19): fixes #170 #187
PR #192 (2025-04-19): Add BMS support and reset offline grid power

### **User description**
## 🔄 Related Issues
Closes #170 #187 #186

___

### **PR Type**
Enhancement, Bug fix

___

### **Description**
- Add BMS support with current and temperature sensors

- Insert inverter model definition sensor

- Adjust PV energy register start offset

- Reset grid inverter power when offline

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Enhancement</strong></td><td><table>
<tr>
  <td>
    <details>
      <summary><strong>enums.py</stron [... cut]
commit b5388d40 (2025-04-19): Add BMS battery temperature (#192)

* fixes #186
* fixes #170 #187
commit 243e5aa8 (2025-04-20): fixes #170 #187
closed 2025-04-19 by Pho3niX90