Pho3niX90/solis_modbus #177: [Feature]: Dropdowns for workmode and battery charge/discharge

Pho3niX90, 2025-04-03:


Pho3niX90, 2025-04-03:
@wawyed tagging you since you wanted this

Pho3niX90, 2025-04-03:
https://github.com/Pho3niX90/solis_modbus/releases/tag/3.2.1

wawyed, 2025-04-03:
Thank you :). Just tested they seem to work correctly. I wanted to ask you if there's an more elegant way to keep the discharge/charge running after the 30min max. Currently I have to send the registry again as below to keep it from turning off:

[image]

Could we either have a toggle that does it internally? Or maybe a button to send the registry again depending on the dropdown option selected?

Thanks for your work!

Pho3niX90, 2025-04-04:
> Thank you :). Just tested they seem to work correctly. I wanted to ask you if there's an more elegant way to keep the discharge/charge running after the 30min max. Currently I have to send the [...]

Perhaps, let me rather ask, why do you want to run it for a prolonged period? Isn't there a workmode that already achieves this?

Whay is your usecase?

wawyed, 2025-04-04:
My use case is my energy provider offers adhoc sessions where I get paid for exporting to the grid and adhoc sesions where is free to import from the grid. So having to setup the timed slots every time is more cumbersome so the force charge discharge is ideal. But those sessions usually last at least 1 hour

closed 2025-04-13 by Pho3niX90