Pho3niX90/solis_modbus #178: [Help]: If I use WaveShare modbus, do I need to create all the sensors myself according to the modbus map?

eqasxar, 2025-04-04:
Do I understand correctly that this integration is prepared to work via WiFi interface? If I use WaveShare modbus, all sensors need to be created by myself according to modbus map. Now I have successfully used Wifi integration, but as in this manual it says lose access to cloud.
I connected from Solis S6 modbus port to Waveshare via ethernet. HA does not receive any data, although I changed the setting from S2_WL_ST to Waveshare and it still remains connected to S2_WL_ST. I restarted HA.

Pho3niX90, 2025-04-05:
No, you do not need to create anything. You will need to make sure your ip is correct, as well as the port.

eqasxar, 2025-04-07:
thanks for the answer, but the problem is that I initially initiated via wifi S2_WL_ST it found the address ip...232, when I set Modify Solis Modbus Settings Connection Type from S2_WL_ST to WAVESHARE, the same ip address remains ...232, waveshare has the address ...200, where can I change it I can't find it?

eqasxar, 2025-04-08:
Everything worked fine via wifi Solis modbus. I switched to Waveshare so I could see the cloud. There are more than 400 entities, but I only see a few numbers. More than one sensor does not show anything. How to fix this to get all the data. IP is correct, Modbus TCP-server baud rate I tried 9600 and 19200. All modbus settings were good and worked with Solis5, I switched to solis S6 integration with Waveshare and I get nothing.

Pho3niX90, 2025-04-09:
You will need to provide debug logs.

eqasxar, 2025-04-10:
I am currently using MGate MB3180, but it is not working. I changed A-B, B-A. Request Slave 1, baud rate 9600

This error originated from a custom integration.

Logger: custom_components.solis_modbus.modbus_controller
Source: custom_components/solis_modbus/modbus_controller.py:148
integration: Solis Modbus (documentation, issues)
First occurred: 01:17:27 (28 occurrences)
Last logged: 01:37:31

Failed to read holding register 43340: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
Failed to read holding register 43483: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
Failed to read holding register 43487: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
Failed to read holding register 43249: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
Failed to read holding register 43291: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request

Pho3niX90, 2025-04-11:
I would need the full debug logs. 

The minimal logs just show that the device isn't responding to holding registrar's, but doesn't indicate if anything else is succeeding.

Pho3niX90, 2025-04-13:
> Hi, it looks like I found a solution after long experiments trying to connect to the third-party RS485 port. I tried every possible method and came to the conclusion that it simply doesn’t work on my inverter. I plugged the Ethernet cable into the Wi-Fi dongle, and now I’m getting data both through Solis Cloud and through Home Assistant. So far, it’s been working for over an hour.

Interesting. So you are using the standard S2 dongle, but just plugged in the ethernet, and both cloud and HA is working?

Pho3niX90, 2025-04-14:
> I sincerely apologize. It was working for a few hours. I was periodically checking Home Assistant or the cloud app and could see both interfaces. Then the cloud disconnected and showed 'LAN mode offline.' As far as I understand, the issue is not with the solis_modbus plugin or Home Assistant. The Solis S6 RS485 interface is unreachable. I tried various Modbus scanners and used different converters from various manufacturers: Waveshare, Moxa MGate MB3180, and ZLAN5143D. The Moxa MGate Manager indicates network errors, even though the cable length is only 0.2 meters.

Try use a new cable, preferably shielded to make sure you don't have any interference. 

As far as the cloud and modbus goes, staright into the normal dongle is known to disconnect the cloud, and it only realizes it's not connefted when there hasn't been any data pushed for 5-15mins. 

The only workaround to have both online is with a waveshare device, but haven't attempted this myself.

closed 2025-04-29 by github-actions[bot]