Pho3niX90/solis_modbus #179: [Enhancement]: S6-EH3P20K-H via Elfin-EW11 RS485 to WiFi  adapter, showing wrong battery charge discharge values

rummin-git, 2025-04-04:
I am using solis_modbus  for S6-EH3P20K-H via Elfin-EW11 RS485 to WiFi  adapter. I am getting battery charge energy values with wrong multipliers.
Yesterday, Today battery charge and discharge values showing 1 kWh instead of 10 kWh.
Total battery charge and discharge values showing 1 kWh instead of 100 kWh.

Pho3niX90, 2025-04-05:
See #178

Pho3niX90, 2025-04-05:
See #168

rummin-git, 2025-04-06:
I choose connection type to S2_WL_ST instead of WAWESHARE, now showing battery charge energy values correctly. Thanks!!!

Also can you explain, what does "Updated to V2 Firmware" tick means?

Pho3niX90, 2025-04-14:
> I choose connection type to S2_WL_ST instead of WAWESHARE, now showing battery charge energy values correctly. Thanks!!!
> 
> Also can you explain, what does "Updated to V2 Firmware" tick means? 

V2 firmware is for inverters with the Grid time of use feature.

closed 2025-04-14 by Pho3niX90