Pho3niX90/solis_modbus #187: [Bug]: Active Power value stays when inverter is offline

sshknoll, 2025-04-15:
Hi folks,

I just noticed that the values of the sensors stay when the inverter goes offline (i.e. at night). That way the last active power reading stays put and HomeAssistant senses PV power all through the night.

Is there anything that can be done about that? I will help if a can, but I am not very good at coding.

Pho3niX90, 2025-04-16:
Which inverter?

sshknoll, 2025-04-16:
Sorry forgot to mention: S5 GR1P 4.6 connected via RS485 to ethernet stick from Solis (WL Stick Datslogger)

Pho3niX90, 2025-04-17:
What value is `Inverter Status` on when this happens? Probably related to #170

sshknoll, 2025-04-17:
Status is 2 before it shuts down completely. I figured out that all the values seem to be stuck to the last value they had when the inverter powered down. The integration does not register the inverter being unavailable. When it powers up in the morning, the readings are back online and update correctly again.

Pho3niX90, 2025-04-17:
> Status is 2 before it shuts down completely. I figured out that all the values seem to be stuck to the last value they had when the inverter powered down. The integration does not register the [...]

Can you confirm the status when it comes back online, perhaps? With this info, I can perhaps code in exceptions. I believe it only changes it status before going offline, without zeroing other values.

sshknoll, 2025-04-18:
The status changes to 3 when it is generating. As far as I know the stati correspond with Appendix 2 in this table.

https://akkudoktor.net/uploads/short-url/kbwIjN8pl8idrCP04UstaxtSmlC.pdf

hbpv, 2025-04-18:
Same here and it always was this way.. In the time intim you Check the latest Update time of an entity value you will Not instantly notice that the Connection is lost.. aspecially with values that mostly Not change in 10-15 seconds ie SOC… this, when noticed to Late, will damage the statistics (Daily) too (loads, pv power)

So values should Not persist if modbus connection is lost but the Entities should Go „Not available“

sshknoll, 2025-04-19:
You are right. Either the integration should check for status or last update longer ago than e.g. two minutes. That way the error margin should still be small.

Pho3niX90, 2025-04-19:
> Same here and it always was this way.. In the time intim you Check the latest Update time of an entity value you will Not instantly notice that the Connection is lost.. aspecially with values that [...]

For this, there is a last successful sensor, changing the sensors to not available will cause more issues with autmations than it will fix, and, will possibly break a lot of current integrations.

sshknoll, 2025-04-19:
Thank you and I am sorry to say that it actually made things worse. After installing 3.3.4 no sensor values updated anymore. Going back to 3.3.3 immediately resolved the issue.

Pho3niX90, 2025-04-19:
@sshknoll I would need logs.

sshknoll, 2025-04-19:
Here you go :)

[home-assistant_solis_modbus_2025-04-19T14-09-13.940Z.log](https://github.com/user-attachments/files/19821473/home-assistant_solis_modbus_2025-04-19T14-09-13.940Z.log)

Pho3niX90, 2025-04-20:
please try https://github.com/Pho3niX90/solis_modbus/releases/tag/3.3.5

sshknoll, 2025-04-20:
Thanks for the update, especially on easter sunday! :) Positive and negative results: The integration is working again as expected, but the sensor values still stick to the last known values they had before the inverter goes offline. The inverter status that you check in the code is 2, but the other sensors are still stuck and show power that is being generated. 
Might it be a solution to zero out the values in a loop when the inverter goes to status 2?

Pho3niX90, 2025-04-21:
> uck and show power that is being generate

Just to better understand, since I cannot test this myself, as I am using a hybrid. 

When inverter goes to status 2, does it still update? Can you perhaps see the debug logs as they will provide the traces of whats happening.

sshknoll, 2025-04-21:
Ok, I re-checked the values again:

The inverter does not necessarily go into status 2 when it goes offline. Sometimes status zero is sensed as well. 

However status 2 simply signals that no power is being generated, other sensors might stay online (grid freqency) and only stop when the inverter really shuts down due to a lack of power.

[image]
[image]
[image]

sshknoll, 2025-04-21:
BTW

"unique": "solis_modbus_inverter_total_load_power",

is used twice in the string_sensors.py and throws an error in HASS log.

sshknoll, 2025-04-22:
I dug into it and was able to include a watchdog to check for timeout. Works for me. I made a fork of the code, please review.

Pho3niX90, 2025-04-22:
> I dug into it and was able to include a watchdog to check for timeout. Works for me. I made a fork of the code, please review.

Perfect, ypu can create a PR and I can review it, and include it into the next release.

sshknoll, 2025-04-22:
> > I dug into it and was able to include a watchdog to check for timeout. Works for me. I made a fork of the code, please review.
> 
> Perfect, ypu can create a PR and I can review it, and include it into the next release.

Great! I will check for functionality and errors again when the sun is down and create a PR.

PR #197 (2025-04-23): Fix unique ID for Load Total Consumption Energy sensor

### **User description**
## 🔄 Related Issues
Closes #187 

Minor change of UID to avoid identical names.

___

### **PR Type**
Bug fix

___

### **Description**
- Rename unique sensor ID for load consumption energy

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Bug fix</strong></td><td><table>
<tr>
  <td>
    <details>
      <summary><strong>string_sensors.py</strong><dd><code>Rename Load Total Consumption Energy sensor unique ID</code>&nbsp; &nbsp; &n [... cut]
PR #196 (2025-04-29): Add sensor watchdog for availability timeout

### **User description**
## 🔄 Related Issues
Closes #187 

## ✅ Testing Steps
1. **Tested With**: Solis S5-GR1P4.6K
2. **Test Results**: Sensors become unavailable after they have not been updated for 10 min. (i.e. when inverter shuts down or Modbus connection was lost)

___

### **PR Type**
Enhancement, Bug fix

___

### **Description**
- Add watchdog for stale sensor updates

- Set sensors unavailable after timeout

- Make timestamps timezone aware

- Refresh availability on update

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><t [... cut]
closed 2025-04-23 by Pho3niX90