Pho3niX90/solis_modbus #191: [Enhancement]: S6-EH3P10K-H-ZP

Dominic-070, 2025-04-17:
Today I made an extension cable and connected a datalogger to my Solis S6-EH3P10K-H-ZP inverter. It turns out that it is not the waveshare where the multipliers differ but the inverter itself. The datalogger also sees the other values. Now it turns out that it is in the inverter itself, you could possibly add an inverter type S6-EH3P10K-H-ZP and apply the adjusted multipliers to it. The waveshare option can be removed then.

[image]

Pho3niX90, 2025-04-17:
What sensors are incorrect, and by how much?

Dominic-070, 2025-04-17:
"register": ['33161', '33162'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.01,  Instead of 1

"register": ['33163'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.01, Instead of 0.1 

"register": ['33164'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.01, Instead of 0.1 

"register": ['33165', '33166'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.01, Instead of 1 

"register": ['33167'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.01, Instead of 0.1

"register": ['33168'], "device_class": SensorDeviceClass.ENERGY, "multiplier": 0.01, Instead of 0.1

Pho3niX90, 2025-04-18:
Those are sensors the waveshare setting affects? Have you enabled/disabled it, restarted home assistant, and give it 15mins to settle? 

Take note, it won't affect history, only new values.

Dominic-070, 2025-04-18:
That's right, these are the Waveshare settings.

I always thought that the Waveshare was the problem of the multipliers or maybe the connection method on the RS485 port.
But yesterday I made an extension cable for my inverter and connected a Solis 
S2-WL-ST datalogger and it also sees the same deviations.

So the problem is not in the Waveshare or in the connection method.

The problem of the multipliers is in the model S6-EH3P10K-H-ZP (which was specially developed for Dutch energy supplier Zonneplan and normally only communicates via their Dongle).

The multipliers work fine when I choose inverter model Waveshare and connection type waveshare. Both via Wavershare adapter and via S2-WL-ST.

My suggestion is only, because the problem is not in waveshare but in S6-EH3P10K-H-ZP, replace `inverter model` from `Waveshare` to `S6-EH3P10K-H-ZP`

Pho3niX90, 2025-04-18:
Have you completely removed the waveshare device? if it's still connected, it will still change the values, as it's reading from the inverter, and proxying the responses to the datalogger and any other connected devices.

S6-EH3P10K-H-ZP uses the same multipliers as the rest of the hybrids

Dominic-070, 2025-04-18:
S6-EH3P10K-H-ZP is a special (Zonneplan) version of the S6-EH3P10K

I had the S2-WL-ST connected in parallel to the waveshare. So Waveshare on the RS485 port and S2-WL-ST on the COM port.

The S6-EH3P10K-H-ZP has an RJ45 as COM where S6-EH3P10 had a 4 pin screw round or USB.
So I made an RJ45 to 4 pin screw.

I added the S2-WL-ST separately as a device with a different IP address than the Waveshare.

Then I disconnected the waveshare, made a transition for the S2-WL-ST from RJ45 (RS485 port to 4 pin screw).

also removed the entire integration and reinstalled it and added the device again.

Conclusion
The deviation is not in Waveshare but specifically in the model variant S6-EH3P10K-H-ZP.

Also the S2-WL-ST datalogger is getting the wrong values from te  S6-EH3P10K-H-ZP. Thats why i know the  S6-EH3P10K-H-ZP is the problem.
Zie screenshot. in Red translations and real values.

[image]

Pho3niX90, 2025-04-18:
Keep in mind the waveshare proxies it internally to the logger and RJ485, regardless of IP of each device. 

I will have a look to add the specific model

Dominic-070, 2025-04-18:
This is what I have tested so far, f there is anything else I need to test, let me know

Waveshare adapter to RJ485 port, Inverter type `S6-EH3P10` --> Wrong values
Waveshare adapter to RJ485 port, Inverter type `Waveshare `--> Right values
S2-WL-ST to RJ485 port, Inverter type `S6-EH3P10` --> Wrong values
S2-WL-ST to RJ485 port, Inverter type `Waveshare` --> Right values
S2-WL-ST to COM port, Inverter type `S6-EH3P10` --> Wrong values
S2-WL-ST to COM port, Inverter type `Waveshare `--> Right values

Pho3niX90, 2025-04-18:
The integration doesn't affect the cloud at all, so doesn't matter what is done here, the cloud would still be incorrect. This is why I say you need to **completely** (not connected to the inverter, nor the datalogger at all, in any way) remove the waveshare, connect the ST2 normally, and see what the values give you. 

But, if the modbus values received by the integration are incorrect, then it will probably be incorrect in the cloud as well, since there are no documentation stating that `S6-EH3P10K-H-Z` has different multipliers. 

If you have indeed completely removed the waveshare device from the system, have you contacted Solis support to see if this is a known issue?

Dominic-070, 2025-04-18:
There is no issue. The S6-EH3P10K-H-Z is not designed for datalogger.
I asked the #75 / issue #74 for adjustment of the values ​​for Waveshare. Now it appears to me that it is not due to Waveshare but to (Firmware on) the model S6-EH3P10K-H-Z.

So the only thing I suggest: Rename Inverter Type `Waveshare` to `S6-EH3P10K-H-Z`.

Pluimvee, 2025-04-24:
Just wanted to add my 2 cents:

- I agree that WAVESHARE is not a SOLIS_INVERTER type. The S6-EH3P10K-H-**ZP** is (AFAIK and understood from Solis) the same model as the S6-EH3P10K-H-**EU** but missing some external connections. I still need to open my ZP model to take a look at the internal circuit to confirm. Meaning I concure with renaming WAVESHARE as inverter type to S6-EH3P10K-H-ZP **if** there is a difference in code logic with the S6-EH3P inverter

- Secondly: the Waveshare Modbus bridge is just one of the possible implementations of a Modbus TCP, I have actually created my own Modbus Bridge using an ESP8266 and ESPHOME. A better name for WAVESHARE is ModBus TCP and I actually question if the connection type actually differs in code logic (dont see any code flow depnding on it) and wonder if this setting can be removed

Pluimvee, 2025-04-24:
Actually, on line 60 in solis_base_sensor there are some adjustements made in the multipliers based on either invertype or connection type

``` py
   def waveshare_adjustment(self):
        """Adjust multiplier if using WAVESHARE and relevant registers are present."""
        if self.controller.inverter_config.type == InverterType.WAVESHARE or self.controller.inverter_config.connection == "WAVESHARE":
            waveshare_registers = {33142, 33161, 33162, 33163, 33164, 33165, 33166, 33167, 33168}
            if waveshare_registers.intersection(self.registrars):
                self.multiplier = 0.01
```
This could be implemented by a ZP flag on top of the S6-EH3P inverter type to prevent code duplications diffferentiating the EU and ZP versions as the diff are expected to be minimal

Pho3niX90, 2025-04-24:
Just to clarify: the "WAVESHARE" inverter implementation is a legacy option and is scheduled for removal, as it has been replaced by the connectionType: WAVESHARE configuration. Because of this, we won't be renaming "waveshare"—it will be deprecated soon.

Regarding the S6-EH3P10K-EH-ZP model, this is the correct direction for new inverter support. However, I still need confirmation from a user not using a WAVESHARE device to verify if the reported values are genuinely incorrect. It's unusual that the mismatched values seem to align closely with what we'd expect from a WAVESHARE device.

@Pluimvee I assume you are using the same inverter? Are you using a waveshare device? If not, are you seeing the same issues?

Pluimvee, 2025-04-24:
I'm indeed not using a waveshare bridge and have the zonneplan inverter S6-EH3P10K-H-ZP

However I do have the integration configured to

  *
Inverter type: S6-EH3P
  *
Connection type: waveshare

I can try tomorrow to verify if changing the Connection type (and restart HA) will give me incorrect results

Pho3niX90, 2025-04-26:
Whilst I agree the waveshare is supposed to only be a bridge, it indeed changes values for some reason. Since I do not have one of these devices to test myself, I need to go on what the community provides when it comes to bugs. 

Initially, I thought it might have been some special inverter software, but the culprit has always been waveshare. 

I have received the newest documentation from Solis, and will investigate if there is any mention towards the ZP inverter, but I highly doubt they would change only the totals, whilst all other solis inverters are consistent across the board. I worry that this might perhaps be a bug in the software, which might have/will be fixed. 

Are your cloud values also incorrect? Can I ask you to get in touch with support for the modbus documentation for your specific inverter? 

If we can confirm from support that this is the correct behavior of this inverter, then we can implement specific model for it, but if it's something that will be fixed/changed, it's pointless.

Pho3niX90, 2025-04-26:
One a side note. 

Why specifically are you using the tcp-rtu bridge? Does this inverter not come with the S2 dongle? Wondering if it's perhaps the dongle itself converting the values, and the bridges/tcp reading the true values?

Pluimvee, 2025-04-26:
Indeed, the ZP inverter does not come with the S2 dongle. I also have no Solis cloud interface.
As my TCP-RTU bridge is pure it can only be that the S2 dongle converts these registers, and the inverter raw values are too high.

Dominic-070, 2025-04-26:
This inverter has no port for a S2. It comes with a Zonneplan dongle.

as said before. i made an adapter to connect the S2 dongle to the com port instead of the dongle and also an adapter to connect the S2 dongle to the RS485. In both cases the S2 dongle gives the same deviation in the multipliers as the waveshare

[image]

Pho3niX90, 2025-04-26:
So this does lead me to believe then that the S2 might be changing the values from the inverter for the totals. It might be a backward compatibility technique since I believe the older inverters perhaps did not have this granularity of the data. This is just an assumption. 

The newest documentation also states that (just using a single register as an example, the rest are the same) 
33169: 1 <-> 1kWh

The connection types are there for users to easily distinguish their connection type
S2 - common dongle
Waveshare - common bridge

So, would a third option here make sense for these type of inverters? Are they region specific?

Pho3niX90, 2025-04-26:
@Dominic-070 Am I understanding correctly that it's branded as Zonneplan? Instead of Solis?

Dominic-070, 2025-04-26:
Yes correct. This inverter is not for sale in any store. Its only available via and installed by Zonneplan. It has the Solis logo at the front but Zonneplan at the manufacturer plate

Pho3niX90, 2025-04-26:
Okay, I will add a third option
- S2
- Waveshare
- **Zonneplan**

Dominic-070, 2025-04-26:
I think EH3P10K-H-ZP should be a invertertype not Zonneplan a connection methode. It dousnt matter if you use wavehare of S2. Same multipliers

Pluimvee, 2025-04-26:
Totally agree with @Dominic-070 
The multipliers are for this inverter and not dependent on the connection method.

Pho3niX90, 2025-04-29:
see if this solves it https://github.com/Pho3niX90/solis_modbus/releases/tag/3.3.6

Dominic-070, 2025-04-29:
Thanks!!! Seems to look good.
Small detail: dont know if this is normal but Inverter model suddenly changed from `EH3P10K-H-ZP` to `S6 3-Phase HV Hybrid (5-10kW)`

[image]

PR #199 (2025-04-29): Add TCP feature support and S6-EH3P10K-H-ZP inverter

### **User description**
## 🔄 Related Issues
Closes #191

___

### **PR Type**
Enhancement

___

### **Description**
- Introduce TCP feature for inverter communications.

- Append TCP feature for WAVESHARE connection models.

- Register new S6-EH3P10K-H-ZP inverter with TCP.

- Refactor sensor adjustment to `tcp_adjustment`.

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Enhancement</strong></td><td><table>
<tr>
  <td>
    <details>
      <summary><stron [... cut]
closed 2025-04-29 by Dominic-070