Pho3niX90/solis_modbus #195: [Bug]: Updating to 2025.4.3 Breakers Modbus connectivity

miketarg, 2025-04-20:
## 🐛 Bug Description
AS above

## 🔄 To Reproduce
Updating

## ✅ Expected Behavior
Working as beofre

## 📸 Screenshots / Logs
This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solis_modbus/modbus_controller.py:178
integration: Solis Modbus (documentation, issues)
First occurred: 6:10:30 PM (18 occurrences)
Last logged: 6:17:04 PM

Failed to connect [Errno 111] Connect call failed ('192.168.0.83', 502)

PLEASE NOTE I HAVE DONE NOTHING BUT UPGARDED THE SYSTEM AND MODBUS NOW IT WONT CONNECT restarted inverter and LAN switch, it wont connect. If I havent provided enough information please can you tell me what to do and I will provide it!

## 🔧 System Information
### 🔹 Home Assistant:
Core
2025.4.3
Supervisor
2025.04.0
Operating System
15.2
Frontend
20250411.0

### 🔹 Integration:
- **Plugin Version**: 3.3.

[image]

- **Device Model**: [e.g., Solis 5G 6kW, Axitec 10kW]
- **Modbus Connection Type?**: Ethernet fixed IP (192.168.0.83)

## ➕ Additional Context
Any other details that might be helpful for debugging.

Pho3niX90, 2025-04-20:
It works fine with 2025.4.3, logs suggest your inverter isn't replying. Restart datalogger and homeassistant.

Without logs I can't debug.

miketarg, 2025-04-20:
It's suddenly working again, it's very strange I haven't done anything, after updating it can take serval restarts to get it to connect!

Thanks.

closed 2025-04-20 by miketarg

Pho3niX90, 2025-04-20:
> It's suddenly working again, it's very strange I haven't done anything, after updating it can take serval restarts to get it to connect!
> 
> Thanks.

This is normal for modbus. Sometimes it hangs up and locks itself for a while. Sometimes, it needs to be restarted. 

So when you experience issues, restart it once, give it 10 minutes, normally it rectifies itself.

miketarg, 2025-04-20:
Ok thanks 🙏🏻