Pho3niX90/solis_modbus #205: [Bug]: 500 Internal Server Error Server got itself in trouble

ChrizZz90, 2025-05-04:
## 🐛 Bug Description
Getting a 500 internal Server error message when trying to add a new device/integration, after selecting Solis Modbus.

## 🔄 To Reproduce
Steps to reproduce the behavior:
1. Go to settings
2. Select integration
3. Click on + add integration
4. Search for Solis Modbus and click on it
5. Error message pops up

## ✅ Expected Behavior
the integration form for a new inverter

## 📸 Screenshots / Logs
`Logger: aiohttp.server
Quelle: /usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py:451
Erstmals aufgetreten: 19:20:18 (1 Vorkommnisse)
Zuletzt protokolliert: 19:20:18

Error handling request from fe80::6cfc:4bd5:8dae:733b
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 182, in post
    return await self._post_impl(request, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 189, in _post_impl
    return await super()._post_impl(request, data)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 92, in _post_impl
    result = self._prepare_result_json(result)
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 209, in _prepare_result_json
    return _prepare_config_flow_result_json(result, super()._prepare_result_json)
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 145, in _prepare_config_flow_result_json
    return prepare_result_json(result)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 51, in _prepare_result_json
    data["data_schema"] = voluptuous_serialize.convert(
                          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        schema, custom_serializer=cv.custom_serializer
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/voluptuous_serialize/__init__.py", line 40, in convert
    pval = convert(value, custom_serializer=custom_serializer)
  File "/usr/local/lib/python3.13/site-packages/voluptuous_serialize/__init__.py", line 58, in convert
    val.update(convert(validator, custom_serializer=custom_serializer))
               ~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/voluptuous_serialize/__init__.py", line 120, in convert
    if issubclass(schema, Enum):
       ~~~~~~~~~~^^^^^^^^^^^^^^
TypeError: issubclass() arg 1 must be a class`

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: core_2025.4.4
- **Installation Method**: Proxmox virtual machine

### 🔹 Integration:
- **Plugin Version**: 3.3.9
- **Device Model**: Solis S6-EH3P12K02-NV-YD-L
- **Modbus Connection Type?**: Wifi
- **Custom Config Entries?**: not sure if relevant and which might affect this behaviour

## ➕ Additional Context
HA is only available in the home network, not available outside of it.

Pho3niX90, 2025-05-04:
please try https://github.com/Pho3niX90/solis_modbus/releases/tag/3.3.10

ChrizZz90, 2025-05-04:
Works now, thanks for your quick update

commit 8742e87f (2025-05-04): fix: config flow, fixes #205
commit a669a0e7 (2025-05-04): fix: config flow, fixes #205
closed 2025-05-04 by ChrizZz90