Pho3niX90/solis_modbus #207: [Enhancement]: Identifying Where Multiple Inverters (Discussion #201)

Dilbertdog, 2025-05-07:
## 🎯 Enhancement Request
Describe how an existing feature can be improved.

## 🔍 Use Case
Explain why this improvement is needed.

## ✅ Expected Improvement
Describe the expected behavior after the enhancement.

## 🛠️ Implementation Suggestions
If you have an idea on how to improve it, add details.

## ➕ Additional Context
Related issues or documentation.

Pho3niX90, 2025-05-09:
Forgot to include the identifiers in the name, it made sense before HA changed hiding of the unique ids

Will include this today

Pho3niX90, 2025-05-09:
see if this resolves it for you https://github.com/Pho3niX90/solis_modbus/releases/tag/v3.3.12

Dilbertdog, 2025-05-10:
Have installed 3.3.12 and restarted HA - deleted both devices and added them back in.
The use of _mas_ and _sla_ (the two identifiers I used) has made it much easier to identify each invertor
eg _sensor.solis_s6_eh1p_mas_apparent_power_ & _sensor.solis_s6_eh1p_sla_apparent_power_

Pho3niX90, 2025-05-10:
Awesome news. Thanks for letting me know

Dilbertdog, 2025-05-11:
Have reopened, just because, not long after this update was installed, and the inverters readded, the PV production stopped, and a little later a Grid Overload alarm happened. I have reset the inverters twice, but this over load alarm is still there. All the data in the inverter control was jumbled up (eg times of charge / discharge, type of use, ability to export etc etc). Have reset these (as far as I know) but the overload alarm is still present. Have reopened, just in case there is a chance that HA wrote to the inverters some duff data. In terms of HA (as I am a novice) all I had done after installing the update and readding the devices, was to change the entities in the power card to show mas and sla, as of course the entity names had changed. It is very possible that there is a separate Solis issue, but best to reopen this, just in case, whilst it is being fixed (hopefully!) by the Solis support team. Thx R

Pho3niX90, 2025-05-11:
No problem, keep me updated

Dilbertdog, 2025-05-19:
https://github.com/Pho3niX90/solis_modbus/issues/207#issuecomment-2869654671
Issue remains open with Solis - they have done one firmware update but this has not fixed the issue whereby the inverters are not performing as per their settings. I will update as soon as they have found a fix. Many thanks

Pho3niX90, 2025-05-22:
> Is there any way that there can be an option to add these devices as “read only” if you wanted to avoid any risk of changing data ? Guess this possibly an “enhancement” request but once we know it is not an issue with the code I’m happy move this over. Many thanks for all your help….

I am busy categorizing settings into groups, to make sure people can select what's available, and what isn't. 

> Is there any way that, when adding a device, it could write back, or amend, inverter and change settings.

The writing of the modbus only happens explicitly when the fields are adjusted, so nothing is automatically written back. 

It would be great if they can highlight what settings were causing the issue, so that I can investigate it as well on my side to see if perhaps I can create a derived status sensor, to make sure all is well.

Dilbertdog, 2025-07-06:
Solis are unable to advise what entities may or may not have been changed when viewing in HA, hence options to download the current state of each entity, and open as read only when looking at the entity fields would appear to be the pragmatic route to deal with this kind of situation. I will revert to the enhancements and this can now remain closed. Many thanks

PR #209 (2025-05-09): Add multi-inverter identification and CI release enhancements

### **User description**
## 🔄 Related Issues
Closes #207

___

### **PR Type**
Enhancement

___

### **Description**
Add device_identification property.
Append identification to entity names.
Overhaul release CI workflow:
  - manual dispatch trigger
  - dynamic semver version bump

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Enhancement</strong></td><td><details><summary>7 files</summary><table>
<tr>
  <td><strong>modbus_controller.py</strong> [... cut]