Pho3niX90/solis_modbus #210: [Help]: Solis-3P6K-4G support

Mustang1966, 2025-05-12:
Hello,

Would it be possible to add support for the Solis-3P6K-4G inverter?
Here is the link to the inverter's specifications:
https://www.solisinverters.com/global/4g_3p_inverter2/34598.html

Thank you in advance!

Pho3niX90, 2025-05-13:
@Mustang1966 it should theoretically work. If it has the S2 dongle, ethernet or waveshare.

Mustang1966, 2025-05-13:
Thanks for the reply! I do have the S2 dongle, currently connected via Ethernet. I've tried setting it up, but it's not working yet — I'm not sure which inverter type or connection method I should select to get proper communication.

Do you know if I should use a specific Modbus configuration or choose a particular model from the list? Or is there something else I might be missing?

Appreciate any tips — happy to test further if needed!

Pho3niX90, 2025-05-14:
> Modbus configuration or choose a particular model from the list? Or is there something else I might be missing?

I have added this one specifically so that we can work from there, try this version https://github.com/Pho3niX90/solis_modbus/releases/tag/v3.3.13

Mustang1966, 2025-05-14:
Thanks a lot! I’m not at home at the moment, so I won’t be able to test it until the weekend. I’ll give it a try then and let you know if it worked. Appreciate your help!

Mustang1966, 2025-05-17:
The integration now seems to be working well. However, there are a few entities that do not display any values—this could be due to them not being supported by the inverter. Additionally, I noticed that the option to shut down or power off the inverter appears to be missing.

Pho3niX90, 2025-05-17:
> The integration now seems to be working well. However, there are a few entities that do not display any values—this could be due to them not being supported by the inverter. Additionally, I noticed that the option to shut down or power off the inverter appears to be missing.

Let me know which registers have no values, if you can run in debug mode as well for 10mins that will tell us why it has no values. 

Then for new entities, just open a separate issue stating which ones you would like added.

Mustang1966, 2025-05-18:
The following entities are currently unavailable in Home Assistant:
- Clock (Hours)
- Clock (Minutes)
- Clock (Seconds)
- Grid Export Total Active Energy
- Grid Import Total Active Energy
- Inverter EPM Firmware Version
- Load Total Consumption Energy
- Model No
- Total Generation Energy
- Total Load Power

[home-assistant_solis_modbus_2025-05-18T17-58-44.563Z.log](https://github.com/user-attachments/files/20274131/home-assistant_solis_modbus_2025-05-18T17-58-44.563Z.log)

closed 2026-10-25 by Pho3niX90