Pho3niX90/solis_modbus #211: [Enhancement]: Use alias only for the sensor unique ID's

Pluimvee, 2025-05-14:
## 🎯 Enhancement Request
Instead of concatenating : solis_[invertype][alias][sensorname]
I would love sensors to get a unique id formatted as: solis_[alias][sensorname]

## 🔍 Use Case
I had to change the inverter type and re-install the component. Because of the other inverter type all sensors got another unique ID and I had to change a huge amount of dashboards, automations, helpers, etc.

## ✅ Expected Improvement
Using the alias for the sensors will prevent getting new ids when changing the inverter type. In my case I could set the alias to the 'old' inverter type to maintain the sensors 

## 🛠️ Implementation Suggestions
The alias could default to inverter_type to be backwards compatible

## ➕ Additional Context
https://github.com/Pho3niX90/solis_modbus/issues/191
https://github.com/Pho3niX90/solis_modbus/discussions/198

Pho3niX90, 2025-05-16:
I will have a look into this.

Pho3niX90, 2025-05-18:
So in regards to the alias. I have updated the translation to the following: "Alias — used when configuring multiple inverters; leave empty if not applicable."

This is only required for multi inverters, to create unique identifiers. If you are only using a single inverter, this alias is pointless.

Pluimvee, 2025-05-18:
When the alias defaults to the inverter_type then it wouldn't break any integrations.

Only for the ones which have already used the alias, but only when they reinstall the component. I do not expect many have already used the alias as the alias was introduced on May 7

Pho3niX90, 2025-05-18:
The alias is meant to be left empty when there is just a single inverter, then it will default to the original method of creating the unique id's. when it's set, it appends it to distinguish between inverters.

the inverter type isn't being used in the creation of the ID's, at least not in code, home assistant, however, appends the model internally to the names, and id.

in bold, is what home assistant adds
example: solis_**[model]**[alias][sensorname]

Pluimvee, 2025-05-18:
The meaning of Alias is to give something an alternative name. What you describe is more like an index, or ID.

While I understand the current alias is meant as ID, I think the alias could support more use cases:
- to give the inverter a unique label/ID as what you descibe
- to be able to change the model, without changing the sensor id's (the use case I have)
- to be able to give the inverter a friendly name like "Nexus" or "roof_top1" 

I'm not sure how these id's are generated in code. I was expecting it to be generated in here
``` yaml
            unique_id=f"{DOMAIN}_{entity['unique']}" if not identification else f"{DOMAIN}_{identification}_{entity['unique']}"
and
            unique_id="{}_{}_{}".format(DOMAIN, identification if identification is not None else controller.host, entity.get("unique", "reserve")),
```

however some sensors dont have the model type in the unique_id's so I think it is possible to leave out the model type when there is an alias given.

``` yaml
switch.solis_rc_force_battery_charge
```

Pho3niX90, 2025-05-18:
> The meaning of Alias is to give something an alternative name. What you describe is more like an index, or ID.
> 
> While I understand the current alias is meant as ID, I think the alias could support more use cases:
> 
> * to give the inverter a unique label/ID as what you descibe
> * to be able to change the model, without changing the sensor id's (the use case I have)
> * to be able to give the inverter a friendly name like "Nexus" or "roof_top1"
> 
> I'm not sure how these id's are generated in code. I was expecting it to be generated in here
> 
>             unique_id=f"{DOMAIN}_{entity['unique']}" if not identification else f"{DOMAIN}_{identification}_{entity['unique']}"
> and
>             unique_id="{}_{}_{}".format(DOMAIN, identification if identification is not None else controller.host, entity.get("unique", "reserve")),
> however some sensors dont have the model type in the unique_id's so I think it is possible to leave out the model type when there is an alias given.
> 
> switch.solis_rc_force_battery_charge

Yes, the implementation is basically an index/id. That is why it's named identification in the code, but alias in translation. I just couldn't think of a better translation to get the point across in such a short manner.

to be honest, I am not even sure if the model can be programmatically excluded from id from the naming convention. 

the only sensors I think would have the model in, would be switches, and time entities, since these aren't programmatically defined, instead, created by HA

closed 2025-05-21 by Pho3niX90