Pho3niX90/solis_modbus #212: [Bug]: Change RC Force charge/discharge power to max of inverter type

Pluimvee, 2025-05-14:
## 🐛 Bug Description
Unable to set the RC Force charge or discharge to max inverter power

## 🔄 To Reproduce
Change the value of
- number.solis_s6_eh3p_rc_force_battery_charge_power
- number.solis_s6_eh3p_rc_force_battery_discharge_power

to above 6000 Watts will return an error

## ✅ Expected Behavior
I expect I can set the power levels to the maximum of the inverter

## 📸 Screenshots / Logs
Some RC values are still default to 6000
[image]

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: core-2025.5.1

### 🔹 Integration:
- **Plugin Version**: 3.3.13
- **Device Model**: S6-EH3P10K-H-ZP
- **Modbus Connection Type?**: TCP
- **Custom Config Entries?**: No

## ➕ Additional Context
Could this be an issue, causing HA not to recompile the code?
[image]

Pluimvee, 2025-05-18:
Trying to resolve this issue by analyzing and chaging the code. Each time I cleaned __ pycache __ and restarted AH

When I change the configuration in _hybrid_sensor.py_ on lines 669, 674, 679 to 10000 the max is still limited to 6000

According to _solis_base_sensor_ at line 165 the config entry was found or else the max would default to 3000
``` yaml
            max_value=entity.get("max", 3000),
```

Changing the default to 10000 in _solis_base_sensor_ at line 93 doesnt help either, so the self-max_value is not None
``` yaml
    def min_max(self):
        if self.min_value is not None and self.max_value is not None:
            return self.min_value, self.max_value
        if self.unit_of_measurement == PERCENTAGE:
            return 0,100
        if self.unit_of_measurement == UnitOfPower.KILO_WATT:
            return 0,6
        if self.unit_of_measurement == UnitOfPower.WATT:
            return 0,10000
```

Concluding
- changing the sensor config entry to 10000 doesnt help
- the config entry is found, or else it would default to 3000
- and the sensor max_value attribute is set to the value 6000

Pluimvee, 2025-05-18:
Are these sensors maybe created as derived sensors in __ init __ at line 184?

``` yaml
    controller._derived_sensors = [
        SolisBaseSensor(
            hass=hass,
            name=entity.get("name"),
            controller=controller,
            registrars=[int(r) for r in entity.get("register", [])],
            state_class=entity.get("state_class", None),
            device_class=entity.get("device_class", None),
            unit_of_measurement=entity.get("unit_of_measurement", None),
            editable=entity.get("editable", False),
            hidden=entity.get("hidden", False),
            multiplier=entity.get("multiplier", 1),
            category=entity.get("category", None),
            identification=entity.get("identification", None),
            unique_id=f"{DOMAIN}_{entity['unique']}" if not identification else f"{DOMAIN}_{identification}_{entity['unique']}"
```
In the above the min / max are not read from configuration

Pluimvee, 2025-05-18:
I found the problem in lines 83-88 of solis_number.py

``` yaml
    async def async_added_to_hass(self) -> None:
        await super().async_added_to_hass()
        state = await self.async_get_last_number_data()
        if state:
            self._attr_native_value = state.native_value
            self.adjust_min_max_step(state.native_min_value, state.native_max_value, state.native_step)

        # 🔥 Register event listener for real-time updates
        self._hass.bus.async_listen(DOMAIN, self.handle_modbus_update)

    def adjust_min_max_step(self, min_wanted: float | None, max_wanted: float | None, step_wanted: float | None):
        # float(43016) & equalization(43017) voltages, and rated capacity(43019)
        if 43016 in self._register or 43017 in self._register or 43019 in self._register:
            installed_battery = cache_get(self.hass, 43009)

            if is_number(installed_battery):
                # only supported with a user defined battery
                if installed_battery != 2 and not self.registry_entry.disabled:
                    self.registry_entry.disabled = True
                else:
                    self.registry_entry.disabled = False

        # if self._attr_native_min_value != min_wanted and min_wanted is not None:
        #     self._attr_native_min_value = min_wanted
        # if self._attr_native_max_value != max_wanted and max_wanted is not None:
        #     self._attr_native_max_value = max_wanted
        # if self._attr_native_step != step_wanted and step_wanted is not None:
        #     self._attr_native_step = step_wanted
```
I had to comment out the adjust of min/max/step. 
I think because the sensor already existed in HA this code is overriding the attributes which are read from new config

Not sure if this code can/may be removed in total

Pho3niX90, 2025-05-18:
> def min_max(self):

I do believe this can probably removed, it's not being used anywhere in the code currently.

Pho3niX90, 2025-05-18:
> I found the problem in lines 83-88 of solis_number_sensor.py
> 
>     async def async_added_to_hass(self) -> None:
>         await super().async_added_to_hass()
>         state = await self.async_get_last_number_data()
>         if state:
>             self._attr_native_value = state.native_value
>             self.adjust_min_max_step(state.native_min_value, state.native_max_value, state.native_step)
> 
>         # 🔥 Register event listener for real-time updates
>         self._hass.bus.async_listen(DOMAIN, self.handle_modbus_update)
> 
>     def adjust_min_max_step(self, min_wanted: float | None, max_wanted: float | None, step_wanted: float | None):
>         # float(43016) & equalization(43017) voltages, and rated capacity(43019)
>         if 43016 in self._register or 43017 in self._register or 43019 in self._register:
>             installed_battery = cache_get(self.hass, 43009)
> 
>             if is_number(installed_battery):
>                 # only supported with a user defined battery
>                 if installed_battery != 2 and not self.registry_entry.disabled:
>                     self.registry_entry.disabled = True
>                 else:
>                     self.registry_entry.disabled = False
> 
>         # if self._attr_native_min_value != min_wanted and min_wanted is not None:
>         #     self._attr_native_min_value = min_wanted
>         # if self._attr_native_max_value != max_wanted and max_wanted is not None:
>         #     self._attr_native_max_value = max_wanted
>         # if self._attr_native_step != step_wanted and step_wanted is not None:
>         #     self._attr_native_step = step_wanted
> I had to comment out the adjust of min/max/step. I think because the sensor already existed in HA this code is overriding the attributes which are read from new config
> 
> Not sure if this code can/may be removed in total

I made some changes in the code pertaining to this. that being said, it could also perhaps be refactored/removed.

Pluimvee, 2025-05-19:
I installed the latest and it still works:

- You removed the max_value, but left the min_value and step to what was initially created during first-time-setup. I also suggest to remove the step and min values, to have these configured each time the inverter, code, config is updated.
# if self._attr_native_max_value != max_wanted and max_wanted is not None:
    #     self._attr_native_max_value = max_wanted

- Both the following methods can be removed
     def min_max(self):
and
    def get_step(self, wanted_step):

- The default step should be set to the mutliplier (this is still an issue in changing the RC Force registers)

- the default 3000 in the following makes no sense. There is a default in the config already. If not given I propose a defualt of None or 0
max_value=entity.get("max", 3000),

Pho3niX90, 2025-05-20:
> I installed the latest and it still works:
> 
> - You removed the max_value, but left the min_value and step to what was initially created during first-time-setup. I suggest to also remove the step and min values, to allow reconfiguring when the code, the config or the inverter is changed.
> ```yaml
> Removed
>  if self._attr_native_max_value != max_wanted and max_wanted is not None:
>     #     self._attr_native_max_value = max_wanted
> 
> Still in 
>         if self._attr_native_min_value != min_wanted and min_wanted is not None:
>             self._attr_native_min_value = min_wanted
>         if self._attr_native_step != step_wanted and step_wanted is not None:
>             self._attr_native_step = step_wanted
> 
> ```
> - Both the following methods can be removed
> ```yaml
>     def min_max(self):
> and
>     def get_step(self, wanted_step):
> ```
> - The default step should be set to the mutliplier (this is still an issue in changing the RC Force registers)
> 
> - the default 3000 in the following makes no sense. There is a default in the config already. If not given I propose a defualt of None or 0
> ```yaml
> max_value=entity.get("max", 3000),
> ```

I do believe the number entities require a default, max and min set, otherwise it throws an error. The defaults are whats in the docs, but gets set soon as modbus receives the real data. 

If you want to do a PR you are most welcome, as I might only get around to it during the weekend.

Pluimvee, 2025-05-22:
Solved
(THere may still be some cleanup)

closed 2025-05-22 by Pluimvee