Pho3niX90/solis_modbus #216: [Bug]: Battery Rated Capacity - wrong multiplicator?

felixkrohn, 2025-05-18:
## 🐛 Bug Description
When I change the value for "Battery Rated Capacity" in the Solis device settings on HA, the value gets set back to the initial/default value (12.6) after a couple seconds, and  I observe the following in the debug logs:

```
2025-05-18 19:27:47.566 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Queued Write Holding Register register = 43019, value = 1000
[...]
2025-05-18 19:27:48.477 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Write Holding Register register = 43019, value = 1000, int_value = 1000: WriteSingleRegisterResponse(dev_id=1, transaction_id=2799, address=43019, count=0, bits=[], registers=[1000], status=1)
2025-05-18 19:27:49.198 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Holding Register 43361: [1800, 0, 0, 0, 0, 0, 100, 0, 0]
[...]
2025-05-18 19:27:58.524 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43009, register 43019 has value 126
```
seeing the value being `1000` when I entered `100` , and it being reset to `126` and displayed as `12.6` it looks to me like it gets multiplied by 10 somewhere.

## 🔄 To Reproduce
Steps to reproduce the behavior:
0. enable debug logging
1. **Go to**: Solis device
2. **Click on**:
3. **Perform action**: set a value in Battery Rated Capacity
4. **Expected result**: value is set
5. **Actual result**:
  - value is reset to last value, debug logs shows value*10
  - trying to set it to "10" instead of "100" yields the following error: 
> Value 10 for number.solis_s6_eh1p_battery_rated_capacity is outside valid range 50.0 - 9999.0."

## 📸 Screenshots / Logs
- log lines: see above

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: 2025.4.2
- **Installation Method**: Docker

### 🔹 Integration:
- **Plugin Version**: v3.3.14
- **Device Model**:  S6-EH1P (1-Phase LV Hybrid Inverter (S6)(Protocol 33) )
- **Modbus Connection Type?**: Ethernet
- **Custom Config Entries?**: No

## ➕ Additional Context
The device is connected by LAN but has no internet access, so I can exclude the possibility that the value gets somehow reset by SolisCloud.

Pho3niX90, 2025-05-18:
Try https://github.com/Pho3niX90/solis_modbus/releases/tag/v3.3.15

Pho3niX90, 2025-05-18:
Note. This setting is only applicable for when user defined battery is selected

felixkrohn, 2025-05-19:
Thanks! the multiplicator setting seems to work as intended now.

closed 2025-05-19 by felixkrohn