Pho3niX90/solis_modbus #217: [Bug]: "Max Charge SOC" does not update

felixkrohn, 2025-05-18:
## 🐛 Bug Description
I'd like to limit my maximum SOC to 90% to prolong battery life. The value is currently set to `100` (default value I guess). The value slider allows the value range `70..100`. When setting it to 90 (or any other value I tried so far) the value gets reset to the initial value (`100`) after a couple  (variable, about 5-15s) seconds.

## 🔄 To Reproduce
Steps to reproduce the behavior:
1. **Go to**: Solis device
2. **Click on**:
3. **Perform action**: slide value ruler to desired value
4. **Expected result**: value remains set permanently
5. **Actual result**: value slider goes back to initial position (100) after a couple seconds

## ✅ Expected Behavior

## 📸 Screenshots / Logs

[image]

[image]

- **Home Assistant logs** (filtered to relevant errors)

> 2025-05-18 19:18:13.806 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Queued Write Holding Register register = 43010, value = 90
> 2025-05-18 19:18:14.881 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] Write Holding Register register = 43010, value = 90, int_value = 90: WriteSingleRegisterResponse(dev_id=1, transaction_id=2222, address=43010, count=0, bits=[], registers=[90], status=1)
> 2025-05-18 19:18:28.524 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] block 43009, register 43010 has value 100

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: 2025.4.2
- **Installation Method**: Docker

### 🔹 Integration:
- **Plugin Version**: v3.3.14
- **Device Model**:  S6-EH1P (1-Phase LV Hybrid Inverter (S6)(Protocol 33) )
- **Modbus Connection Type?**: Ethernet
- **Custom Config Entries?**: No

## ➕ Additional Context
This might be the same or a similar issue as in #97.
The device is connected by LAN but has no internet access, so I can exclude the possibility that the value gets somehow reset by SolisCloud.

I read in other issues that "If it reverts back to the original value, it means writing failed". How can I help debug this further? It's the first time I have to deal with modbus.

Pho3niX90, 2025-05-18:
Have you selected the user defined battery?

felixkrohn, 2025-05-19:
No, I chose LV Dyness (4).
I just _briefly_ tried setting it to "user define" and the same happened. but as this is a thing I prefer testing after sunset in order not to have any (wrong) charging currents on the battery, I will try it again more thoroughly tonight.

That being said, is there a reason for this? Limiting the SOC to 90% to me is an inverter feature that all batteries should support anyways?

Pho3niX90, 2025-05-19:
> No, I chose LV Dyness (4). I just _briefly_ tried setting it to "user define" and the same happened. but as this is a thing I prefer testing after sunset in order not to have any (wrong) charging [...]

I am not sure for the reason behind this, it's just noted in the documentation.  Would be why the inverter is not accepting the changes. 

I do agree, that this should be settable by the user.

felixkrohn, 2025-05-19:
> noted in the documentation

Can you maybe share (a link to) that documentation? I haven't found any deeply technical docs about those inverters. I'd like to understand which values I have to manually set when using "user defined" battery, as right now it always switches back to the model I had set initially after about 1-2 minutes - I assume this is due to some invalid values in fields that have to be set manually.
Also, I might try to open a ticket/ask my installer to open a ticket once I understand the settings that are exposed through modbus a bit better to ask about the "Max Charge SOC" feature.

Pho3niX90, 2025-05-21:
There is a document linked here https://github.com/Pho3niX90/solis_modbus/issues/107

closed 2025-05-21 by Pho3niX90