Pho3niX90/solis_modbus #218: [Bug]: Trying to get my head arround the working mode

Pluimvee, 2025-05-20:
Each time I try to put the inverter to standby it starts to charge. So I dug into all the workmodes and settings trying to find out what was telling it to start charging. And I got a bit lost, or even confused

In _select.py_ (line 40) there is a workmode specified
```yaml
         "register": 43110,
            "name": "Work Mode",
            "entities": [
                {"bit_position": 0, "name": "Self-Use Mode", "work_mode": (0,1,2,6,11)},
                {"bit_position": 1, "name": "Time Of Use Mode", "work_mode": (0,1,2)},
                {"bit_position": 2, "name": "OFF-Grid Mode", "work_mode": (0,1,2)},
                {"bit_position": 6, "name": "Feed In Priority Mode", "work_mode": (0,6,11)},
                {"bit_position": 11, "name": "Solis Peak-shaving Mode", "work_mode": (0,4,6,11)},
            ]
```

Looking at the code in _solis_select_entity_ (line 74) I see that the work_mode first disables the bits of the other modes, before enabling the bit_position of the enabled mode. Some modes can not be enabled together.

``` yaml
        if bit_position is not None:
            if work_mode is not None:
                for wbit in work_mode:
                    current_register_value = set_bit(current_register_value, wbit, False)

            new_register_value: int = set_bit(current_register_value, bit_position, True)
``` 

Now instead of using the select, you can also switch the modes individually.  In the switch.py (line 29) however, there is another 'bit clearing' definition. For example the "Time Of Use Mode" and "OFF-Grid Mode" don't clean the other modes.

``` yaml
                "register": 43110,
                "entities": [
                    {"bit_position": 0, "name": "Solis Self-Use Mode", "work_mode": (0,6,11)},
                    {"bit_position": 1, "name": "Solis Time Of Use Mode"},
                    {"bit_position": 2, "name": "Solis OFF-Grid Mode"},
                    {"bit_position": 3, "name": "Solis Battery Wakeup Switch"},
                    {"bit_position": 4, "name": "Solis Reserve Battery Mode", "work_mode": (4,11)},
                    {"bit_position": 5, "name": "Solis Allow Grid To Charge The Battery"},
                    {"bit_position": 6, "name": "Solis Feed In Priority Mode", "work_mode": (0,6,11)},
                    {"bit_position": 7, "name": "Solis Batt OVC"},
                    {"bit_position": 8, "name": "Solis Battery Forcecharge Peakshaving"},
                    {"bit_position": 9, "name": "Solis Battery current correction"},
                    {"bit_position": 10, "name": "Solis Battery healing mode"},
                    {"bit_position": 11, "name": "Solis Peak-shaving Mode", "work_mode": (0,4,6,11)},
                ]
```

Looking at the functional code which uses the above definition, in _solis_binary_sensor_ (line 96), I see the same initial clearing of the work_mode bits (when the mode is enabled) after which the new mode bit is set
 
``` yaml
            if value is True and self._work_mode is not None:
                for wbit in self._work_mode:
                    current_register_value = set_bit(current_register_value, wbit, False)

            new_register_value: int = set_bit(current_register_value, self._bit_position, value)
```

Which of the above is the correct mode_clearing?

What is the diff between
- Storage control Switch value (register 43110) where the above bits are written to, and
- Storage Control Switching Value (register 33132) which is changing a bit later, but seems to leave bit 0 always enabled (Self-Use)

Pho3niX90, 2025-05-21:
43110 is the holding register, to change values, whilst 33132 reports on it. I did make a note to remove 33132 completely, since it missed some critical information in the documentation. This seems to be the case for most registers when it comes to report vs holding.

The reason why self user stays enabled, could be due to conflicting modes, and it falls back to the original, the holding registers aren't read often, and typical are set when changing the value, and the modbus responds with success.

Great find!

Pluimvee, 2025-05-21:
ps. The work mode select does not have a 'None' to reflect when all mode bits are cleared. Is this correct and the possible cause for the Self-Use mode to automaticly turn on again?

My inverter tends to have the value 3 in register 33132, meaning both Self Use and Time to Use modes are enabled. However according to your bitmaps these modes can't be combined.

Pluimvee, 2025-05-21:
> I should probably think about how to better handle this. Do you have suggestions?

I think the select entity is a nice approach but would suggest to remove the following until the swiches are stabilized. The below is resulting in duplicaitons regarding this topic

``` yaml
        {
            "register": 43110,
            "name": "Work Mode",
            "entities": [
                {"bit_position": 0, "name": "Self-Use Mode", "work_mode": (0,1,2,6,11)},
                {"bit_position": 1, "name": "Time Of Use Mode", "work_mode": (0,1,2)},
                {"bit_position": 2, "name": "OFF-Grid Mode", "work_mode": (0,1,2)},
                {"bit_position": 6, "name": "Feed In Priority Mode", "work_mode": (0,6,11)},
                {"bit_position": 11, "name": "Solis Peak-shaving Mode", "work_mode": (0,4,6,11)},
            ]
        }
```

I would also rethink if you want to follow up on your note below. I suggest to have the binary_sensors (R/O) based on 33132 and the switches (R/W) on 43110

```yaml
## Remove 33132 in next update if this works (43110)
```

For the rest I think your code is perfect and nice for the mutually exclusive versions (switchin off one when turing on the other). Instead of _on_mode_ you could call this _deselect_ (as you will deselect the ones identified)

The others are flags with a precondition to have another being on or off. So could could call this "condition: (1)"

Looking at the table I see three modes which can keep the current logic
- Self-Use 
- Feed-In Priority 
- Off-Grid 

TOU and Peak shaving can be switched on/off when mode is either
- Self-Use 
- Feed-In Priority 

The TOU and PeakShaving have a precondition but should not change mode as we cant tell which one to use.

Regarding the tooltips: Maybe we can use HA for this when trying to switch something on?
[image]

Pluimvee, 2025-05-21:
Hmmmm
Feed-in mode is bit 6 isnt it?

Bit 3 as you mentioned above is the Wakeup Switch

PR #219 (2025-05-21): Align switch & select work_mode logic and names

### **User description**
## 🔄 Related Issues
Closes #218

___

### **PR Type**
Bug fix, Enhancement

___

### **Description**
- Standardize `work_mode` bit tuples uniformly

- Remove `Solis` prefix from entity names

- Harmonize generator and grid switch names

- Reformat entity definitions for readability

___

### **Changes walkthrough** 📝
<table><thead><tr><th></th><th align="left">Relevant files</th></tr></thead><tbody><tr><td><strong>Enhancement</strong></td><td><table>
<tr>
  <td>
    <details>
      <summary><strong>switch.py</strong><dd> [... cut]
closed 2025-05-22 by Pho3niX90