Pho3niX90/solis_modbus #224: [Bug]: S6-EH3P10 K-H-EU/S2-WL-ST starting with v3.3.16 some values don´t get updates

hplx-1, 2025-05-23:
## 🐛 Bug Description
A clear and concise description of what the bug is.
On my S6-EH3P10 K-H-EU/S2-WL-ST starting with v3.3.16 some values from the "Solar card Setup" don´t get updates
- Solis S6-EH3P Today Battery Charge Energy
- Solis S6-EH3P Today Battery Dicharge Energy

Solis S6-EH3P Today Energy Consumption (didn´t work 2 days, 19.05.+20.05.)

I reverted to v3.3.15 which solved the problem.

## 🔄 To Reproduce
Steps to reproduce the behavior:
1. **Go to**: [e.g., Energy Dashboard, Developer Tools → States]
2. **Click on**: [e.g., a specific sensor, switch, or setting]
3. **Perform action**: [e.g., toggling a switch, checking logs]
4. **Expected result**: [e.g., switch should turn on, value should update]
5. **Actual result**: [e.g., state doesn't change, wrong reading]

## ✅ Expected Behavior
A clear and concise description of what you expected to happen.

## 📸 Screenshots / Logs
If applicable, add screenshots or logs to help explain your problem.
- **Home Assistant logs** (filtered to relevant errors)

## 🔧 System Information
### 🔹 Home Assistant:
- **Version**: [2025.5.2]
- **Installation Method**: [e.g., Home Assistant OS, Docker, Supervised]

### 🔹 Integration:
- **Plugin Version**: [e.g., 1.2.3]
- **Device Model**: [e.g., Solis 5G 6kW, Axitec 10kW]
- **Modbus Connection Type?**: [e.g., Wifi, Ethernet, Waveshare]
- **Custom Config Entries?**: [Yes/No]

## ➕ Additional Context
Any other details that might be helpful for debugging.

Pho3niX90, 2025-05-23:
Can you confirm which inverter model you have chosen? Did you perhaps choose the S6-EH3P10K-EH-ZP? The ZP model selection would cause this behavior

hplx-1, 2025-05-24:
[image]

SOLAR-RAIDER, 2025-05-24:
have the same Problem with my S6-EH3P15K02

Pluimvee, 2025-05-26:
This code
``` yaml
       # adjust some registers for S6-EH3P10K-H-ZP inverter type
        elif self.controller.inverter_config.model in ("S6-EH3P10K-H-ZP"):
            registers = {33142, 33161, 33162, 33163, 33164, 33165, 33166, 33167, 33168}
            if registers.intersection(self.registrars):
                self.multiplier = 0.01
```
needs to be changed to
```yaml
       # adjust some registers for S6-EH3P10K-H-ZP inverter type
       elif self.controller.inverter_config.model == "S6-EH3P10K-H-ZP":
            registers = {33142, 33161, 33162, 33163, 33164, 33165, 33166, 33167, 33168}
            if registers.intersection(self.registrars):
                self.multiplier = 0.01
```

Sorry, my bad

PR #229 (2025-05-27): Fix model condition for S6-EH3P10K-H-ZP multiplier logic

## 🔄 Related Issues
Closes #224

## ✅ Testing Steps
Code review

## ➕ Additional Notes
The Tuple with one element ("S6-EH3P10K-H-ZP") is not a tuple—it's just a string. As such the substring S6-EH3P is also resolving into True and the multipler adjustment is executed