Pho3niX90/solis_modbus #231: [Enhancement]: max. charge/discharge current S6-EH3P10 K-H-EU

hplx-1, 2025-05-28:
## 🎯 Enhancement Request
[Enhancement]: max. charge/discharge current S6-EH3P10 K-H-EU #202 old issue

max. charge/discharge current expected for S6-EH3P10 K-H-EU: 0 - 50 (A)
for some dedicated actions I need values between 0 - 18.5

Still not possible to set values in Home Assistant:
"Value 0 for number.solis_s6_eh3p_battery_max_charge_current is outside valid range 5 - 340."

## ✅ Expected Improvement
valid range 0 - 340

## 🛠️ Implementation Suggestions
allow values starting with 0

## ➕ Additional Context
Screenshot 2025-05-28 101401

hplx-1, 2025-06-09:
Logger: homeassistant.components.websocket_api.http.connection
Quelle: components/websocket_api/commands.py:292
Integration: Home Assistant WebSocket API (Dokumentation, Probleme)
Erstmals aufgetreten: 03:16:31 (2 Vorkommnisse)
Zuletzt protokolliert: 03:17:09

[140440025596448] Value 2.0 for number.solis_s6_eh3p_battery_max_charge_current is outside valid range 5 - 340
[140440025596448] Value 1.0 for number.solis_s6_eh3p_battery_max_charge_current is outside valid range 5 - 340

jonathan-ek, 2025-06-24:
I had the same problem but with another entity.
I managed to work around the issue by commenting out line 58 here [sensors/solis_number_sensor.py#L58](https://github.com/Pho3niX90/solis_modbus/blob/b6206e95bcd07035d97407341cc5d335529744df/custom_components/solis_modbus/sensors/solis_number_sensor.py#L58) and restart home assistant to get the updated limits and not just the stored limits.

I'm not sure if removing that call permanently have negative effects, maybe @Pho3niX90 knows? Based on the commit message I would guess it has something to do with improving number entity restoring.

hplx-1, 2025-06-29:
> I had the same problem but with another entity. I managed to work around the issue by commenting out line 58 here [...]

Thank you.
Finaly it is possible to manage my S6-EH3P10 K-H-EU (part off).
Unfortunately, you can´t manage discharge current (number.solis_s6_eh3p_battery_max_discharge_current), still looked.

hplx-1, 2025-07-22:
> I had the same problem but with another entity. I managed to work around the issue by commenting out line 58 here [...]

I am trying to set "battery_max_discharge_current" below 5.
Any hints were to find the code, commenting out line 58 didn´t help for "battery_max_discharge_current".

hplx-1, 2025-08-01:
I´m using solar power to charge my car. If a cloud passes by and the minimum power is not available I drain some power from battery or grid (for 300 sec). I want to use max. 0,5 kWh from battery (regular household load) by setting battery_max_discharge_current to 1.5.

julmnk, 2025-08-02:
> I´m using solar power to charge my car. If a cloud passes by and the minimum power is not available I drain some power from battery or grid (for 300 sec). I want to use max. 0,5 kWh from battery (regular household load) by setting battery_max_discharge_current to 1.4 A equal to 0,5 kWh.

So, you go at HA to Developer tools enter search entity , there you will find exact name and actual value. After under Settings -> Automation & Scenes write a trigger question what trigger would be and why don't you want to set 1.4A at inverter's e.g. Solis battery settings.... so if you don't want to set default 1.4A in inverters Battery settings as max discharge you can create an automation but if you want help you shall explain what trigger (-s) you would like to use to set 1.4A .... I am sharing what kind of complex automation I wrote to control my Solis S6 inverter which each 15 min is checking batter voltage and if
 condittion is voltage <51.5V als oand if Feed In prio. is on also and if time between 11 and 21 o'clock switch to Self Use mode:

alias: 3.1_Solis S6 Switch to Self-Use Mode if battery_voltage <51.5V 
description: "" 
triggers: 
  - minutes: /15 
    trigger: time_pattern 
conditions: 
  - condition: numeric_state 
    entity_id: sensor.solis_s6_eh3p_s6_eh3p15k02_nv_l_battery_voltage 
    below: 51.5 
  - condition: and 
    conditions: 
      - condition: state 
        entity_id: switch.feed_in_priority_mode 
        state: "on" 
  - condition: and 
    conditions: 
      - condition: time 
        after: '11:00:00' 
        before: '21:00:00' 
actions: 
  - action: switch.turn_on 
    target: 
      entity_id: switch.self_use_mode 
    data: {} 
mode: single 

[image]

Pho3niX90, 2025-08-05:
> I am trying to set "battery_max_discharge_current" below 5. Any hints were to find the code, commenting out line 58 didn´t help for "battery_max_discharge_current".

Afaik the lowest the S6 can go is 5, if you want to deny battery drain a solution such as what @julmnk  provided might be best

hplx-1, 2025-08-05:
Solis S6-EH3P10K-H-EU can do 0,1A with a high voltage battery.

(0-18.5 with my Pylontech Force H2)

 

For now I´m using an other way:

my wallbox is managing the current (number.solis_s6_eh3p_battery_max_discharge_current) drawn from “house battery”, set to 1.4 A.

 

I´m still looking for a workaround in home assistant, in an other step I want to manage the way my heat pump drains the battery.

commit c57e848d (2025-08-26): fix: adjust battery discharge current min value and comment out unused method call, fixes #231
closed 2025-08-26 by Pho3niX90