Pho3niX90/solis_modbus #234: [Integration]: S6-EH3P(8-15)K02-NV-YD-L

ChrizZz90, 2025-05-29:
## 🔗 Inverter Model Details
- **Brand**: Solis
- **Model Number**: S6-EH3P(8-15)K02-NV-YD-L
- **Connection Type**: Modbus TCP / maybe also Waveshare

## 🔍 Why is this Important?
Because I have one haha - if you can tell me how to support, happy to be able to help...

## 📄 Documentation / Modbus Registers
- https://www.solisinverters.com/global/energy_storage_inverters22/S6-EH3P(8-15)K_gl.html
- where can I get the Modbus register information?

## ➕ Additional Context
Currently I use S6-EH1P as the base for the integration, most of the things work, but especially the battery status is missing

ChrizZz90, 2025-05-29:
So battery SOC is 0% but in the SolisCloud App it's currently 79%. And the battery SOC is also continuously 0%.

Pho3niX90, 2025-05-30:
> So battery SOC is 0% but in the SolisCloud App it's currently 79%. And the battery SOC is also continuously 0%.

This inverter model doesn't have special requirements when it comes to battery SOC, so it should be working fine. Are you looking at the correct sensor?

Perhaps just enable debug logging for 2 minutes and upload the logs so that I can investigate

Also, S6-EH3P might be more fitting than S6-EH1P

Pho3niX90, 2025-05-30:
Strange, it seems that your inverter is indeed reporting 0 for SOC
 block 33132, register 33139 has value 0

Have you selected/configured the battery correctly? Is it a "48V-LiBat-LV"? For comms from your BMS, it should be "Lithium Battery LV(RS485)" at the very least. The SOC is what the BMS reports. 

Which battery do you have installed?

Pho3niX90, 2025-05-30:
> strange...how can the SolisCloud app show a value? I thought it is configured correctly. It's a basen green 51.2V LiFePo4 battery https://www.basengreen.com/de/product/eu-stock-51-2v-300ah-15kwh-wall-mounted-lifepo4-battery-pack-lifepo4-for-solar-energy-storage/
> 
> But I use CAN protokoll and connection instead of RS485...

The name is just generic, RS485 would be can as well. 

In the solis app, what does it state the battery there? I believe the solis app would probably guesstimate the percentage based on either the voltage or the coulomb count.

Pho3niX90, 2025-06-01:
If the comms cable is connected to the inverter, you can just select the one with comms. That should be sufficient enough.
The basegreen has both can and rs485

Pho3niX90, 2025-06-01:
The batter naming in the integration might not be 100% spot on, as it's just from the documentation, which isn't always up to date. 

I double checked in my app, and mine is set on "Lithium battery LV", which does have comms for CAN BMS, and RS485 is "Lithium Battery LV(RS485", here is a screenshot for comparison

[image]

closed 2025-06-05 by Pho3niX90

ChrizZz90, 2025-06-05:
works now...there was an issue with the communication cable plugged in into the wrong connector. Therefore S6-EH3P(8-15)K02-NV-YD-L works aswell with the plugin