Pho3niX90/solis_modbus #243: [Enhancement]: Ability to Print All The Sensor Information

Dilbertdog, 2025-06-07:
## 🎯 Enhancement Request
Provide a print function to enable all the entity information (name and status / setting) for the selected device to be saved at the point the print is made.

## 🔍 Use Case
It is easy to accidently make a change via modbus (and not realise) which can happen particularly when you are looking at infomation on a phone or scrolling down. If an issue occurs then it would be easier to refer back to the configuration of the device at that point in time to identify what has changed, so the entity can be reset

## ✅ Expected Improvement
Via use of print button, be able to print all the entity information (capturing as much of the entity name as possible) and settings, which can then be saved as a pdf or hard copy.

## 🛠️ Implementation Suggestions
If you have an idea on how to improve it, add details.

## ➕ Additional Context
Currently trying to print the device and entity information only provides one screen of print in Home Assistant (this is via Google Chrome). Windows 11 Snipping Tool does not have the ability to scroll the whole of the webpage to capture the information.
In my case, Solis are attempting to get the system working again, but I am unable to provide them with what the settings were before the system was corrupted. A print of the set up at that point in time, would enable a comparison to be made between the "then" and "now" situation for each entity.

Pho3niX90, 2025-06-07:
I am not sure if this is possible with HA, I would have to investigate.

Dilbertdog, 2025-07-06:
If it is not possible to print the current state of each entity, is it possible that the current states for each entity in a device could be downloaded as a csv file, so a record could be kept of device settings at that point in time? Many thanks for your continued efforts with this.

closed 2025-08-04 by github-actions[bot]