Pho3niX90/solis_modbus #258: Sensors does not update

RusI4n, 2025-07-03:
Most sensors of Solis S6-EH3P has no update after HA update to 2025-7-0

2025-07-03 10:42:03.490 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️ Received None for register 43173 - 43190, fro (192.168.1.230.1), skipping.
2025-07-03 10:42:03.490 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 43361 starting for (192.168.1.230.1)
2025-07-03 10:42:03.631 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:05.055 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:07.725 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 9 but got 10, IGNORING.
2025-07-03 10:42:08.631 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:10.055 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:12.510 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️(192.168.1.230.1) Skipping FAST update: A previous instance is still running
2025-07-03 10:42:12.615 ERROR (MainThread) [pymodbus.logging] No response received after 5 retries, continue with next request
2025-07-03 10:42:12.615 ERROR (MainThread) [custom_components.solis_modbus.modbus_controller] (192.168.1.230.1) Failed to read input register 33530: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request
2025-07-03 10:42:12.616 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️ Received None for register 33530 - 33535, fro (192.168.1.230.1), skipping.
2025-07-03 10:42:12.616 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ✅ FAST update completed in 270.1227s
2025-07-03 10:42:12.619 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 33000 starting for (192.168.1.230.1)
2025-07-03 10:42:12.755 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:12.953 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️(192.168.1.230.1) Skipping FAST update: A previous instance is still running
2025-07-03 10:42:13.631 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:15.063 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:17.756 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:18.631 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:18.632 WARNING (Thread-1 (_monitor)) [homeassistant.util.logging] Module pymodbus.logging is logging too frequently. 200 messages since last count
2025-07-03 10:42:18.638 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️(192.168.1.230.1) Skipping FAST update: A previous instance is still running
2025-07-03 10:42:19.956 ERROR (MainThread) [custom_components.solis_modbus.modbus_controller] (192.168.1.230.1) Failed to read input register 33000: Modbus Error: [Input/Output] ERROR: No response received of the last requests (default: retries+3), CLOSING CONNECTION.
2025-07-03 10:42:19.956 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️ Received None for register 33000 - 33019, fro (192.168.1.230.1), skipping.
2025-07-03 10:42:19.957 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] Group 35000 starting for (192.168.1.230.1)
2025-07-03 10:42:19.957 DEBUG (MainThread) [custom_components.solis_modbus.modbus_controller] (192.168.1.230.1) Connecting to Modbus TCP...
2025-07-03 10:42:20.114 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 1 but got 2, IGNORING.
2025-07-03 10:42:22.755 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:23.631 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 10 but got 11, IGNORING.
2025-07-03 10:42:25.114 WARNING (MainThread) [pymodbus.logging] ERROR: expected transaction 1 but got 2, IGNORING.
2025-07-03 10:42:25.264 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Clock (Hours) in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.266 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Clock (Minutes) in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.268 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Clock (Seconds) in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.270 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Total Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.274 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Current Month Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.277 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Last Month Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.278 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Today Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.278 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Yesterday Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.278 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV This Year Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.279 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Last Year Energy Generation in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.279 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Max Inverter Current in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.279 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Battery Temperature (BMS) in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.280 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Voltage 1 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.280 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Current 1 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.280 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Voltage 2 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.280 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Current 2 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.281 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Voltage 3 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.281 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Current 3 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.281 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Voltage 4 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.281 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Current 4 in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.282 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Total PV Power in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.282 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor Alarm code data in over 10 minutes. Setting to 0.
2025-07-03 10:42:25.282 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor PV Bus Voltage in over 10 minutes. Setting to 0.
[... cut ...]
2025-07-03 10:42:25.432 WARNING (MainThread) [custom_components.solis_modbus.sensors.solis_sensor] ⚠️ No Modbus update for sensor RC Timeout in over 10 minutes. Setting to 0.
2025-07-03 10:42:27.503 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️(192.168.1.230.1) Skipping SLOW update: A previous instance is still running
2025-07-03 10:42:27.505 DEBUG (MainThread) [custom_components.solis_modbus.data_retrieval] ⚠️(192.168.1.230.1) Skipping NORMAL update: A previous instance is still running

josu98, 2025-07-06:
I have the same problem after upgrading HA earlier this week. I have alsa registered an issue regarding this.

josu98, 2025-07-14:
Anything new around this problem?

hplx-1, 2025-07-14:
No.

Works fine for me.

josu98, 2025-07-21:
> > Anything new around this problem?
> 
> I fixed it by disabling Solis Cloud integration: delete Solis Modbus integration, disable Solis Cloud, than add Solis Modbus again (there were Solis Modbus integration update during this procedure). After that I enabled Solis Cloud and everything works now.

Ok thanks. 
How/where did you disable Solis Cloud?

RusI4n, 2025-07-21:
> Ok thanks. How/where did you disable Solis Cloud?

I just disabled integration

plandregan, 2025-08-12:
I have just found with 1 inverter, it was ok, add 2 or 3, different IPs, but same inverter type, only the last to reload works