Pho3niX90/solis_modbus #264: [Enhancement]: Populate the Parallell entitites with data

lordblc, 2025-07-15:
## 🎯 Enhancement Request
From my Waveshare connected inverter, i have these 3 entitites:
sensor.solis_s6_eh3p10k_h_zp_hjemme_1_485_parallel_inverter_ac_current
sensor.solis_s6_eh3p10k_h_zp_hjemme_1_485_parallel_inverter_ac_voltage
sensor.solis_s6_eh3p10k_h_zp_hjemme_1_485_parallel_inverter_power

None of them have any data in them despite the inverters being connected up:

[image]

[image]

## 🔍 Use Case
It seems i cannot get any communication over ModBus from the second inverter when they are synced up. It would therefore be practical to get this data through the one link that does work.
Being new to the entire world of Solis inverters, i don't know if this is expected behaviour or not. Can anyone with a similar setup chime in? Did you get ModBus working alongside the parallell connection on all connected inverters? If so, what did you do?

The model inverter i have is S6-EH3P10K-H-LV on both. Currently only the master has a battery connected.

I understand i might be alone in this configuration as of today, but i will gladly work with you to make it happen so others can enjoy this feature as well. :)
I love the integration for what it is today. You've done a really god job this far.

## ✅ Expected Improvement
Not sure what to put in this field.. 

## 🛠️ Implementation Suggestions
-

## ➕ Additional Context
-

lordblc, 2025-08-12:
bump..
hoping the lack of response is just summer holiday..

Pho3niX90, 2026-04-13:
I will be having a second inverter installed today to observe the parallel inverters