Pho3niX90/solis_modbus #268: 3.3.5 <> 3.3.28

hbpv, 2025-08-05:
Hey @Pho3niX90 

still appriciate your work here! I'm stuck on 3.3.5 while this fixed all noticeable errors and the modbus connection reliability (mostly). Also theres no feature missing (for me) @3.3.5.

Nevertheless you made some huge steps towards to x.28, so my question for this:

- Is there any major braking change, especially for long term (day/month/overall) statistics that i may burn jumping to the latest version?

Also, could it fix the still occuring errors, see beneath:

1st:

Logger: pymodbus.logging
Quelle: runner.py:154
Erstmals aufgetreten: 22. Juli 2025 um 08:45:45 (**4611 Vorkommnisse**)
Zuletzt protokolliert: 08:46:53

Failed to connect [Errno 111] Connect call failed ('192.168.0.26', 502)
Unable to decode frame Modbus Error: Unknown response 59
Failed to connect
ERROR: received pdu without a corresponding request, IGNORING
Unable to decode frame Modbus Error: [Input/Output] byte_count 38 > length of packet 23

2nd:

Logger: homeassistant.components.sensor.recorder
Quelle: components/sensor/recorder.py:366
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 23. Juli 2025 um 00:05:10 (5 Vorkommnisse)
Zuletzt protokolliert: 4. August 2025 um 00:00:10

Entity sensor.solis_s6_solis_yesterday_energy_imported_from_grid from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 1.1 (1.2) with last_updated set to 2025-07-22T22:00:20.371524+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
Entity sensor.solis_s6_solis_pv_yesterday_energy_generation from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 50.7 (52.7) with last_updated set to 2025-07-25T21:59:57.308721+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
Entity sensor.solis_s6_solis_yesterday_energy_consumption from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 13.0 (14.3) with last_updated set to 2025-07-28T22:00:05.320534+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
Entity sensor.solis_s6_solis_yesterday_energy_fed_into_grid from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 28.6 (31.6) with last_updated set to 2025-07-31T22:00:12.923594+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues
Entity sensor.solis_s6_solis_yesterday_battery_charge_energy from integration solis_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 5.9 (6.1) with last_updated set to 2025-08-03T21:59:53.320904+00:00. Please create a bug report at https://github.com/Pho3niX90/solis_modbus/issues

3rd:

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.solis_modbus.modbus_controller
Quelle: custom_components/solis_modbus/modbus_controller.py:148
Integration: Solis Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 23. Juli 2025 um 07:11:42 (3 Vorkommnisse)
Zuletzt protokolliert: 2. August 2025 um 10:31:38

Failed to read holding register 43010: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.26:502]
Failed to read holding register 43010: Modbus Error: [Input/Output] No response received after 5 retries, continue with next request

Pho3niX90, 2025-08-05:
Hi 

**1st/3rd** This isn't an issue with the plugin. The is solely connection related, either from the inverter, router etc. It also makes use of the internal pymodbus, the are susceptible to errors as well 

**2nd:** 
If this is an issue in the latest version, a bug needs to be opened for it.

closed 2025-08-05 by Pho3niX90

Pho3niX90, 2025-08-05:
Also, the latest release was about a month ago, to see stability you can check the issues overview. But there hasn't been issues, and when there are, a hotfix is shortly followed.