Pho3niX90/solis_modbus #274: [Bug]: Sensors unavailable after update to HA 2025.9.0

DieterC1984, 2025-09-04:
## 🐛 Bug Description
Tons of sensors not available anymore since HA upgrade to 2025.9.0
Version 3.3.12

scoof, 2025-09-04:
Cause:
`2025-09-04 09:18:31.533 ERROR (MainThread) [custom_components.solis_modbus.modbus_controller] (192.168.86.107.1) Failed to read input register 33580: ModbusClientMixin.read_input_registers() got an unexpected keyword argument 'slave'`
I fixed it by changing line 194 of modbus_controller.py to
`result = await self.client.read_input_registers(address=register, count=count, device_id=self.slave)`
I will submit a PR later today if I can figure out the root cause of this change, unless someone else does it before me.

Arnoudd, 2025-09-04:
> Cause: `2025-09-04 09:18:31.533 ERROR (MainThread) [custom_components.solis_modbus.modbus_controller] (192.168.86.107.1) Failed to read input register 33580: ModbusClientMixin.read_input_registers() [...]

Same issue here, and this ^^^ fixed it for me too. Thanks!

0rangutan, 2025-09-04:
Also fixed using this change, thanks @scoof!

Arnoudd, 2025-09-04:
Looks like I was a bit to quick. The fix by @scoof partially fixed it for me but some sensors did go to 'unavavailable' again. And I got the same error on some write registers too.

So I had to make the same change in  three more places in the modbus_controller.py (change 'slave=self.slave' to 'device_id=self.slave') .

omenukas, 2025-09-04:
> Looks like I was a bit to quick. The fix by [@scoof](https://github.com/scoof) partially fixed it for me but some sensors did go to 'unavavailable' again. And I got the same error on some write [...]

I also noticed that many sensors still do not work after changing the line, so I returned the system to the previous version

Arnoudd, 2025-09-04:
> I also noticed that many sensors still do not work after changing the line, so I returned the system to the previous version

Same here, after a while it seems to be losing connection. 
Now I remember why I normally never install a new *.0 version update.

renebiemans, 2025-09-05:
Same issue here. Can you fix already be downloaded maybe as beta?

omenukas, 2025-09-05:
> Same issue here. Can you fix already be downloaded maybe as beta?

The fact that there is no Beta version yet

0rangutan, 2025-09-05:
It has worked 100% fine for me after making the 4 line changes.
I'm sure that @Pho3niX90 will be online again soon to release an update,

Ian-Johnston, 2025-09-05:
Thanks Pho3niX90.

I am on an old version of this (V1.6.2)......can I just replace the files or do I need to uninstall the integration and start again?
Back when I installed I configured for a Solis S6, and modded the code a bit to fit my RAI-3K-48ES-5G which is battery only (no solar).

Ian.

0rangutan, 2025-09-05:
I'd update to the newly released 3.3.32 Ian.
In the setup for it you can now choose whether you have solar, battery and/or generator so this should support your setup.