Pho3niX90/solis_modbus #281: [Feature]: DC Power sensor.

Milh0us3, 2025-09-07:
## 🚀 Feature Request
Is there possible to add DC Power sensor to see which MPPT generating W. Same as in SolisCloud shows. Now only shows, Amperes and Voltage and I missing Watts.

There I add and image how shows on SolisCloud power. It would be great to have here too.

[image]

julmnk, 2025-09-07:
[image]

julmnk, 2025-09-07:
[image]

Pho3niX90, 2025-09-15:
> Unfortunately but mine is not hybrid. It’s S5-GR3P15K

Got it. S5 and S6 are the same, but each has different types, such as Hybrid, Grid/String, Energy. 

Yours is the Grid/String type inverter. 

Will have a look to add to add the dc sensors.